coolsnowwolf/lede #12390: 自编译的R5S设备固件，开机无法启动

lysgwl, 2024-08-06:
### 详细叙述

新买的设备R5s，然后为其编译最新的固件，tf卡启动设备，设备无法启动，只有电源灯亮。然而使用官方的friendlywrt固件SD卡启动就没有问题，什么原因造成的。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

friendlywrt R5s

### 详细日志

[0_build.txt](https://github.com/user-attachments/files/16510994/0_build.txt)

lysgwl, 2024-08-06:
> 记得这机器启动默认emmc，至于你说的友善固件sd可以启动，这就很神奇了

可以的，emmc只要没有引导程序，sd卡就可以启动了。按Mask键也可以强制sd强制启动。friendlywrt固件引导启动是木问题的。

lysgwl, 2024-08-06:
> 刷进emmc试试

试过了，不管是SD刷，还是先刷，自编译的固件刷进去，就起不来。官方的friendlywrt和istoreos是可以用的，但这两个有点臃肿，不太习惯。

olprc, 2024-08-06:
刚我编译X86，也启动不了。刷回上个固件了。。。

lysgwl, 2024-08-07:
> 我的r5s没有问题，起码能启动，所以不是这里的问题

分享下你的固件呗，试试看行不行？

lysgwl, 2024-08-07:
> 刚我编译X86，也启动不了。刷回上个固件了。。。

那就不知道了呀，在讨论arm架构的r5s设备呢。

lysgwl, 2024-08-07:
> > > 我的r5s没有问题，起码能启动，所以不是这里的问题
> > 
> > 
> > 分享下你的固件呗，试试看行不行？
> 
> 去恩山看看吧

哦，你也是下的呀。

qingtian110, 2024-08-18:
一个只为 NanoPi R5S 打造的固件  https://r5s.cooluc.com/

lysgwl, 2024-08-19:
> 一个只为 NanoPi R5S 打造的固件 https://r5s.cooluc.com/

你说的这个固件，前几天就测试过了，同样是无法启动。我用的是SD卡启动模式，emmc启动没有测试。

qingtian110, 2024-08-19:
> > 一个只为 NanoPi R5S 打造的固件 https://r5s.cooluc.com/
> 
> 你说的这个固件，前几天就测试过了，同样是无法启动。我用的是SD卡启动模式，emmc启动没有测试。

说明你的 emmc 存在 uboot 且未被清空。你应该使用强制SD 启动（长按 Mask 按钮 + 通电）

lysgwl, 2024-08-21:
> > > 一个只为 NanoPi R5S 打造的固件 https://r5s.cooluc.com/
> > 
> > 
> > 你说的这个固件，前几天就测试过了，同样是无法启动。我用的是SD卡启动模式，emmc启动没有测试。
> 
> 说明你的 emmc 存在 uboot 且未被清空。你应该使用强制SD 启动（长按 Mask 按钮 + 通电）

emmc的分区和uboot也都已经清空了，强制SD启动也尝试过。现已经emmc安装了iStoreOS ,就没再尝试了。

coolsnowwolf, 2026-09-05:
已检查所附 GitHub Actions 构建日志：该次构建最终成功生成 `friendlyarm_nanopi-r5s` 的 squashfs/ext4 sysupgrade 镜像及 manifest，日志中没有 LEDE 编译失败。当前 issue 仅描述设备无法启动，没有串口/U-Boot 输出、kernel log、镜像校验和或启动介质写入方式，因此无法判断是 U-Boot、启动介质、设备树还是硬件问题。

目前不能据此修改代码或关闭 issue。请补充：串口全量启动日志、使用的镜像文件名及 SHA256、U-Boot 版本/启动命令，以及是否按 R5S 的 mask 按键方式强制从 SD 启动。

closed 2026-09-05 by coolsnowwolf