coolsnowwolf/lede #12392: 自编译的R4S设备固件，开机后lan口配置似乎不对

banana121, 2024-08-07:
### 详细叙述

自编译的R4S设备固件，可以正常开机，开机后lan口配置似乎不对，PC插lan口无法获取ip，指示灯也不闪烁。wan看起来正常，但连接不上lan口，无法确认系统启动的具体状态。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

nanopi R4S

### 详细日志

无，编译安装正常。

yanxin152133, 2024-08-07:
wan口设置成lan口也是这样吗

banana121, 2024-08-08:
> wan口设置成lan口也是这样吗
厉害，我都没想到，交换了一下，至少能登录路由了。初步差了一下ifconfig，没看到eth1

root@OpenWrt:~# ifconfig
br-lan    Link encap:Ethernet  HWaddr 04:91:62:F0:4B:86
          inet addr:192.168.1.1  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fdb8:8d96:36e4::1/60 Scope:Global
          inet6 addr: fe80::691:62ff:fef0:4b86/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2431 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1715 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:239416 (233.8 KiB)  TX bytes:601150 (587.0 KiB)

eth0      Link encap:Ethernet  HWaddr 04:91:62:F0:4B:86
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2484 errors:0 dropped:12 overruns:0 frame:0
          TX packets:1851 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:275944 (269.4 KiB)  TX bytes:607934 (593.6 KiB)
          Interrupt:43

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:274 errors:0 dropped:0 overruns:0 frame:0
          TX packets:274 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:21193 (20.6 KiB)  TX bytes:21193 (20.6 KiB)

banana121, 2024-08-08:
> wan口设置成lan口也是这样吗

应该是默认没有勾选网络驱动，我把驱动编译进去就好了，多谢！

closed 2024-08-08 by banana121

banana121, 2025-03-20:
> > > wan口设置成lan口也是这样吗
> > 
> > 
> > 应该是默认没有勾选网络驱动，我把驱动编译进去就好了，多谢！
> 
> 遇到了一样的问题，请问你是勾选了哪个驱动，就好了呀？

时间有点久了，好像是r8169的驱动。