coolsnowwolf/lede #12394: OrangePi_Zero3的i2c-0总线地址被占用

QYKing233, 2024-08-08:
### 详细叙述

当我在 26pin 引脚中的 VCC GND SDA SCL 连接 oled 显示模块后运行 i2cdetect -y 0 输出如下：
$ i2cdetect -y 0      
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- UU -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 

查看 sun50i-h616.dtsi 发现 0x36 的地址被 PMIC 使用，所以还能通过硬件 i2c 驱动 oled 模块吗，或者使用 GPIO 的其他引脚模拟 i2c

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

Orangepi Zero3

### 详细日志

无

QYKing233, 2024-08-08:
是 sun50i-h618-orangepi-zero3.dts 并非 sun50i-h616.dtsi

aiamadeus, 2024-08-08:
[image]
为什么要去管 i2c0，板子上只有 i2c3 啊。。
```
--- a/arch/arm64/boot/dts/allwinner/sun50i-h616-orangepi-zero.dtsi
+++ b/arch/arm64/boot/dts/allwinner/sun50i-h616-orangepi-zero.dtsi
@@ -72,6 +72,12 @@ &emac0 {
 	status = "okay";
 };
 
+&i2c3 {
+	pinctrl-names = "default";
+	pinctrl-0 = <&i2c3_ph_pins>;
+	status = "okay";
+};
+
 &mdio0 {
 	ext_rgmii_phy: ethernet-phy@1 {
 		compatible = "ethernet-phy-ieee802.3-c22";
```

QYKing233, 2024-08-08:
成功驱动了，谢谢大佬。

QYKing233, 2024-08-08:
成功驱动了，谢谢大佬。

closed 2024-08-08 by QYKing233

Linn-Cui, 2024-09-08:
> 成功驱动了，谢谢大佬。

求指导，也遇到了同样问题，，我是openwrt