coolsnowwolf/lede #12395: 缺少clang后编译报错失败

jzhou404, 2024-08-08:
### 详细叙述

ubuntu-23.04-live-server-amd64 环境编译

[1/295] CXX src/base/command_line.o
FAILED: src/base/command_line.o
clang++ -MMD -MF src/base/command_line.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++20 -Wrange-loop-analysis -Wextra-semi-stmt -c ../src/base/command_line.cc -o src/base/command_line.o
/bin/sh: 1: clang++: not found
ninja: build stopped: subcommand failed.

 sudo apt install clang 后问题解决。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

J3160

### 详细日志

[1/295] CXX src/base/command_line.o
FAILED: src/base/command_line.o
clang++ -MMD -MF src/base/command_line.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++20 -Wrange-loop-analysis -Wextra-semi-stmt -c ../src/base/command_line.cc -o src/base/command_line.o
/bin/sh: 1: clang++: not found
ninja: build stopped: subcommand failed.

es202517, 2024-08-29:
你安装clang之后没报unknown warning option '-Wno-cast-function-type'; did you mean '-Wno-bad-function-cast'? [-Werror,-Wunknown-warning-option]这个错误么？

loopback123, 2025-04-28:
> 你安装clang之后没报unknown warning option '-Wno-cast-function-type'; did you mean '-Wno-bad-function-cast'? [-Werror,-Wunknown-warning-option]这个错误么？

这个错误我遇到了, 是我ubuntu虚拟机clang版本过低, 卸载clang, 安装clang-14  llvm-14, 设置环境: PATH=/usr/lib/llvm-14/bin:$PATH 
 source /etc/profile 一下就解决了

coolsnowwolf, 2026-09-05:
日志的直接原因是主机未安装 `clang++`（`/bin/sh: clang++: not found`），并非 LEDE 源码错误。当前 README 的依赖列表已明确包含 `clang`、`llvm`；评论中另一个 warning 也已确认是 Ubuntu 自带 clang 版本过低，安装 clang/llvm-14 后即可解决。

请按 README 安装完整依赖，或使用受支持的 Debian/Ubuntu LTS 环境。本 issue 已有明确解决方法，关闭。

closed 2026-09-05 by coolsnowwolf