coolsnowwolf/lede #12397: rax3000m运行一段时间WAN口掉IP信息，手动关闭重启接口不生效，需要重启恢复。

yanzhx7, 2024-08-09:
### 详细叙述

rax3000m运行一段时间WAN口掉IP信息，手动关闭重启接口不生效，需要重启恢复。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

rax3000m

### 详细日志

[内核日志.txt](https://github.com/user-attachments/files/16556539/default.txt)
[系统日志.txt](https://github.com/user-attachments/files/16556541/default.txt)
[image]
[image]

closed 2024-08-15 by yanzhx7

Rem1x-2019, 2024-09-22:
朋友，我也是这样的问题，无论是光猫拨号还是桥接。。都会这样，
请问你解决了吗。。

Rem1x-2019, 2024-09-22:
> ### 详细叙述
> rax3000m运行一段时间WAN口掉IP信息，手动关闭重启接口不生效，需要重启恢复。
> 
> ### 重复 issue
> * [x]  没有类似的 issue
> 
> ### 具体型号
> rax3000m
> 
> ### 详细日志
> ![1](https://private-user-images.githubusercontent.com/169524030/356477357-07b0a073-c2ca-4533-9a5d-361ab4ca05ec.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY5ODk3NzIsIm5iZiI6MTcyNjk4OTQ3MiwicGF0aCI6Ii8xNjk1MjQwMzAvMzU2NDc3MzU3LTA3YjBhMDczLWMyY2EtNDUzMy05YTVkLTM2MWFiNGNhMDVlYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTIyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyMlQwNzE3NTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05MzVhZTYwOGY1YzM4OTMzNTk3YTEzODI1ZjM2ZDNiNjUzMTJkOTg1MGVlMjQxNzc4MzczNzM5YTQ0NGRmMjUxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.b2Fp70U-V0opF8NOEl7NIZmWJGZjc4bYDLQ7w9M27dw) ![Snipaste_2024-08-09_11-49-32](https://private-user-images.githubusercontent.com/169524030/356477367-2232af8f-31f9-4f8f-98ef-717806449246.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY5ODk3NzIsIm5iZiI6MTcyNjk4OTQ3MiwicGF0aCI6Ii8xNjk1MjQwMzAvMzU2NDc3MzY3LTIyMzJhZjhmLTMxZjktNGY4Zi05OGVmLTcxNzgwNjQ0OTI0Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTIyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyMlQwNzE3NTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zZDUxOGMzYjE5ZmZjZWVjYmE5NmYwYWFlODBkOWI5YzI1ODk5MzE1Mjg1ZTkzNDNhNDEyODM4ZGEwZDEwOWNmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.T-2YqD3tz0987Zk5kHlj39T-AlWVCpEIrhIBG0sqYXY)
> 
> 系统日志 10点左右eth1掉线 Fri Aug 9 10:00:50 2024 kern.warn kernel: [35478.256836] ------------[ cut here ]------------ Fri Aug 9 10:00:50 2024 kern.info kernel: [35478.261458] NETDEV WATCHDOG: eth1 (mtk_soc_eth): transmit queue 1 timed out Fri Aug 9 10:00:50 2024 kern.warn kernel: [35478.268447] WARNING: CPU: 0 PID: 0 at dev_watchdog+0x23c/0x24c Fri Aug 9 10:00:51 2024 kern.debug kernel: [35478.274281] Modules linked in: pppoe ppp_async pppox ppp_mppe ppp_generic mt7915e mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FULLCONENAT xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_bbr slhc nf_tproxy_ipv6 nf_tproxy_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp Fri Aug 9 10:00:51 2024 kern.debug kernel: [35478.274489] nf_conntrack_amanda nf_conncount macvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables compat asn1_decoder crypto_safexcel ntfs3 act_connmark sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb sit tunnel4 ip_tunnel tun nls_utf8 nls_iso8859_1 nls_cp437 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_arm64 sha1_ce sha1_generic seqiv md5 des_generic libdes Fri Aug 9 10:00:51 2024 kern.debug kernel: [35478.361591] authencesn authenc arc4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug vfat fat exfat usbcore usb_common Fri Aug 9 10:00:51 2024 kern.debug kernel: [35478.462421] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 6.1.100 #0 Fri Aug 9 10:00:51 2024 kern.debug kernel: [35478.468416] Hardware name: CMCC RAX3000M (NAND version) (DT) Fri Aug 9 10:00:51 2024 kern.debug kernel: [35478.474061] pstate: 40400005 (nZcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--) Fri Aug 9 10:00:51 2024 kern.debug kernel: [35478.481010] pc : dev_watchdog+0x23c/0x24c Fri Aug 9 10:00:51 2024 kern.debug kernel: [35478.485012] lr : dev_watchdog+0x23c/0x24c Fri Aug 9 10:00:51 2024 kern.debug kernel: [35478.489014] sp : ffffffc008003e20 Fri Aug 9 10:00:51 2024 kern.debug kernel: [35478.492318] x29: ffffffc008003e20 x28: ffffffc008b369c0 x27: ffffffc008003ef0 Fri Aug 9 10:00:51 2024 kern.debug kernel: [35478.499448] x26: 00000000000000e0 x25: 0000000000000000 x24: dead000000000122 Fri Aug 9 10:00:51 2024 kern.debug kernel: [35478.506574] x23: 0000000000000000 x22: ffffffc008b36000 x21: 0000000000000001 Fri Aug 9 10:00:51 2024 kern.debug kernel: [35478.513700] x20: ffffff8000ac6000 x19: ffffff8000ac6440 x18: 000000000000013c Fri Aug 9 10:00:51 2024 kern.debug ke [... cut ...]  [39873.312217] mtk_soc_eth 15100000.ethernet eth1: Link is Down [39876.431503] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx [40446.208375] mtk_soc_eth 15100000.ethernet eth1: Link is Down [40455.566076] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
> 
> 全部日志： [内核日志.txt](https://github.com/user-attachments/files/16556566/default.txt) [系统日志.txt](https://github.com/user-attachments/files/16556568/default.txt)

我已经解决了，换个固件就好了，，。