coolsnowwolf/lede #12401: 连续10天编译失败求大神

Chaychen-TM, 2024-08-10:
### 详细叙述

编译失败

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

乌班图  20.04

### 详细日志

Created filesystem with 4149/524288 inodes and 92042/2097152 blocks
rm -f /home/chay/Desktop/lede/build_dir/target-x86_64_musl/linux-x86_64/generic-kernel.bin
cp /home/chay/Desktop/lede/build_dir/target-x86_64_musl/linux-x86_64/bzImage /home/chay/Desktop/lede/build_dir/target-x86_64_musl/linux-x86_64/generic-kernel.bin
cp /home/chay/Desktop/lede/build_dir/target-x86_64_musl/linux-x86_64/generic-kernel.bin /home/chay/Desktop/lede/bin/targets/x86/64/openwrt-x86-64-generic-kernel.bin
[ -f /home/chay/Desktop/lede/build_dir/target-x86_64_musl/linux-x86_64/generic-kernel.bin -a -f /home/chay/Desktop/lede/build_dir/target-x86_64_musl/linux-x86_64/root.squashfs ]
dd if=/home/chay/Desktop/lede/build_dir/target-x86_64_musl/linux-x86_64/root.squashfs >> /home/chay/Desktop/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz
117521+1 records in
117521+1 records out
60170843 bytes (60 MB, 57 MiB) copied, 0.986877 s, 61.0 MB/s
dd if=/home/chay/Desktop/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz of=/home/chay/Desktop/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz.new bs=8589934592 conv=sync
dd: memory exhausted by input buffer of size 8589934592 bytes (8.0 GiB)
make[5]: *** [Makefile:164: /home/chay/Desktop/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz] Error 1
make[5]: Leaving directory '/home/chay/Desktop/lede/target/linux/x86/image'
make[4]: *** [Makefile:26: install] Error 2
make[4]: Leaving directory '/home/chay/Desktop/lede/target/linux/x86'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/chay/Desktop/lede/target/linux'
time: target/linux/install#358.30#29.75#196.09
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/home/chay/Desktop/lede'
make[1]: *** [target/Makefile:24: /home/chay/Desktop/lede/staging_dir/target-x86_64_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/chay/Desktop/lede'
make: *** [/home/chay/Desktop/lede/include/toplevel.mk:231: world] Error 2

hcym, 2024-08-10:
一个是梯子正常，第二个先用最小配置或者官方原版的最简配置，编译成了就没有心理障碍，以后就顺便了
熟能生巧，24.04也可以试试

QYKing233, 2024-08-10:
你的系统有多大内存？

QYKing233, 2024-08-10:
dd: memory exhausted by input buffer of size 8589934592 bytes (8.0 GiB)
从这里看是写盘的时候内存不足。

jzhou404, 2024-08-11:
你的固件已经编译成功了，看这里：
dd if=/home/chay/Desktop/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz of=/home/chay/Desktop/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz.new bs=8589934592 conv=sync

问题是你的5G内存不够，升级一下内存
dd: memory exhausted by input buffer of size 8589934592 bytes (8.0 GiB)

closed 2024-08-13 by coolsnowwolf