coolsnowwolf/lede #12403: 编译失败

cnkids, 2024-08-11:
### 详细叙述

摸不着头脑，之前没有出现过类似的问题

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86_64

### 详细日志

WARNING: If bootstrapping with this script fails, it may be due to an
WARNING: incompatible installed `libtool.m4' being pulled in to
WARNING: `aclocal.m4'.  The best way to work around such a problem is to
WARNING: uninstall your system libtool files, or failing that, overwrite
WARNING: them with all m4 file as shipped with this distribution (except
WARNING: `lt~obsolete.m4').  After that, retry this bootstrap.
make[4]: Entering directory '/home/free/lede/build_dir/host/libtool-2.4.2'
/home/free/lede/staging_dir/host/bin/autom4te --language=m4sh -B libltdl/config libltdl/config/ltmain.m4sh > libltdl/config/ltmain.in
sed -e s,@MACRO_VERSION\@,2.4.2,g -e s,@MACRO_REVISION\@,1.3337,g -e s,@MACRO_SERIAL\@,,g -e s,@PACKAGE\@,libtool,g -e s,@PACKAGE_BUGREPORT\@,bug-libtool@gnu.org,g -e s,@PACKAGE_URL\@,http://www.gnu.org/software/libtool/,g -e s,@PACKAGE_NAME\@,GNU Libtool,g -e s,@package_revision\@,1.3337,g -e s,@PACKAGE_STRING\@,GNU Libtool 2.4.2,g -e s,@PACKAGE_TARNAME\@,libtool,g -e s,@PACKAGE_VERSION\@,2.4.2,g -e s,@TIMESTAMP\@,,g -e s,@VERSION\@,2.4.2,g ./libltdl/config/ltmain.in > ./libltdl/config/ltmain.sh
sed -e s,@MACRO_VERSION\@,2.4.2,g -e s,@MACRO_REVISION\@,1.3337,g -e s,@MACRO_SERIAL\@,3337,g -e s,@PACKAGE\@,libtool,g -e s,@PACKAGE_BUGREPORT\@,bug-libtool@gnu.org,g -e s,@PACKAGE_URL\@,http://www.gnu.org/software/libtool/,g -e s,@PACKAGE_NAME\@,GNU Libtool,g -e s,@package_revision\@,1.3337,g -e s,@PACKAGE_STRING\@,GNU Libtool 2.4.2,g -e s,@PACKAGE_TARNAME\@,libtool,g -e s,@PACKAGE_VERSION\@,2.4.2,g -e s,@TIMESTAMP\@,,g -e s,@VERSION\@,2.4.2,g ./libltdl/m4/ltversion.in > ./libltdl/m4/ltversion.m4
make[4]: 'libtoolize.in' is up to date.
make[4]: 'tests/defs.in' is up to date.
make[4]: 'tests/package.m4' is up to date.
make[4]: 'tests/testsuite' is up to date.
make[4]: 'libltdl/Makefile.am' is up to date.
make[4]: 'doc/notes.txt' is up to date.
make[4]: Leaving directory '/home/free/lede/build_dir/host/libtool-2.4.2'
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: /home/free/lede/staging_dir/host/bin/aclocal --force -I libltdl/m4
autoreconf: configure.ac: tracing
autoreconf: running: /home/free/lede/build_dir/host/libtool-2.4.2/libltdl/config/libtoolize --copy --force --ltdl
libtoolize: Bootstrap detected, no files installed.
Segmentation fault
autoreconf: /home/free/lede/staging_dir/host/bin/aclocal failed with exit status: 139
make[3]: *** [Makefile:54: /home/free/lede/build_dir/host/libtool-2.4.2/.prepared6df808cbd63b6a32a9dda48c4e8ee158_6664517399ebbbc92a37c5bb081b5c53] Error 139
make[3]: Leaving directory '/home/free/lede/tools/libtool'
time: tools/libtool/compile#2.68#0.20#3.84
    ERROR: tools/libtool failed to build.
make[2]: *** [tools/Makefile:221: tools/libtool/compile] Error 1
make[2]: Leaving directory '/home/free/lede'
make[1]: *** [tools/Makefile:217: /home/free/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnyyynynyyyyyyynyynnynyynnynnyyyyyyyyyyyyyyyyynynnyyyyyyyy] Error 2
make[1]: Leaving directory '/home/free/lede'
make: *** [/home/free/lede/include/toplevel.mk:231: world] Error 2

yanxin152133, 2024-08-12:
https://github.com/openwrt/openwrt/issues/12565

coolsnowwolf, 2026-09-05:
日志的首个失败点是旧版 GNU libtool 2.4.2 在 `aclocal` 阶段段错误（exit 139）。当前 `tools/libtool/Makefile` 已升级至 2.5.4，提交 `bf4f6db117a` 同步了上游版本及 bootstrap 处理，已不再使用日志中的 2.4.2 构建路径。

该 issue 没有提供当前提交或新的复现日志，现有问题已被工具链更新覆盖，无需新增补丁，关闭 issue。

closed 2026-09-05 by coolsnowwolf