coolsnowwolf/lede #12405: Radxa E20C 编译失败 uboot-rk35xx failed to build

Matrix-CN, 2024-08-14:
### 详细叙述

云编译显示 
ERROR: package/boot/uboot-rk35xx failed to build (build variant: rk3528-evb).

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

E20C

### 详细日志

make[4]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rockchip-rk3528-evb/uboot-rockchip-2023-07-25-32640b0a'
  CHK     include/config/uboot.release
  CHK     include/generated/version_autogenerated.h
  CHK     include/generated/timestamp_autogenerated.h
  CHK     include/generated/generic-asm-offsets.h
  CHK     include/generated/asm-offsets.h
start=$(aarch64-openwrt-linux-musl-nm u-boot | grep __rel_dyn_start | cut -f 1 -d ' '); end=$(aarch64-openwrt-linux-musl-nm u-boot | grep __rel_dyn_end | cut -f 1 -d ' '); tools/relocate-rela u-boot-nodtb.bin 0x00200000 $start $end
  COPY    u-boot.dtb
./"arch/arm/mach-rockchip/make_fit_atf.sh" \
arch/arm/dts/rk3528-evb.dtb > u-boot.its
/usr/bin/env: 'python2': No such file or directory
ls: cannot access 'bl31_0x*.bin': No such file or directory
  MKIMAGE u-boot.itb
Error: u-boot.its:46.29-30 syntax error
FATAL ERROR: Unable to parse input tree
./tools/mkimage: Can't read u-boot.itb.tmp: Invalid argument
make[4]: *** [Makefile:1056: u-boot.itb] Error 255
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rockchip-rk3528-evb/uboot-rockchip-2023-07-25-32640b0a'
make[3]: *** [Makefile:77: /workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rockchip-rk3528-evb/uboot-rockchip-2023-07-25-32640b0a/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/boot/uboot-rk35xx'
time: package/boot/uboot-rk35xx/rk3528-evb/compile#1.20#0.59#1.73
    ERROR: package/boot/uboot-rk35xx failed to build (build variant: rk3528-evb).
make[2]: *** [package/Makefile:116: package/boot/uboot-rk35xx/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2

dragon888888888888, 2024-08-14:
[image]
可能是因为这里变成多选且不可取消吧

coolsnowwolf, 2024-08-14:
/usr/bin/env: 'python2': No such file or directory

缺 python2 的包

closed 2024-08-14 by Matrix-CN