coolsnowwolf/lede #12406: 编译失败

Wyatt323, 2024-08-14:
### 详细叙述

编译时老是这里失败

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

Configuring kmod-igc.
Configuring kmod-bnx2.
Configuring urngd.
Configuring luci-i18n-firewall-zh-cn.
Configuring 6to4.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
Configuring luci-i18n-minidlna-en.
make[2]: Leaving directory '/workdir/openwrt'
 * pkg_hash_check_unresolved: cannot find dependency luci-theme-infinityfreedom for luci
 * pkg_hash_fetch_best_installation_candidate: Packages for luci found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for default-settings:
 * 	luci-theme-infinityfreedom
 * opkg_install_cmd: Cannot install package default-settings.
 * resolve_conffiles: Existing conffile /workdir/openwrt/build_dir/target-x86_64_musl/root-x86/etc/config/socat is different from the conffile in the new package. The new conffile will be placed at /workdir/openwrt/build_dir/target-x86_64_musl/root-x86/etc/config/socat-opkg.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[1]: *** [package/Makefile:111: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2

lanruihua026, 2024-08-15:
pkg_hash_check_unresolved: cannot find dependency luci-theme-infinityfreedom for luci
luci-theme-infinityfreedom这个主题哈希值错误，openwrt无法找到对应的依赖，请向软件源作者仓库反馈，**取消勾选该主题**

Wyatt323, 2024-08-15:
[image]

又有新的错了

coolsnowwolf, 2026-09-05:
日志的首个致命错误是：`default-settings` 依赖 `luci-theme-infinityfreedom`，但当前 feeds/源码中不存在该主题，导致 `pkg_hash_check_unresolved` 和 `package/install` 失败。后续 `socat` conffile 提示不是导致失败的原因。

该依赖属于用户配置/第三方主题选择，不是 x86 编译器或 LEDE 核心包缺陷。取消 `luci-theme-infinityfreedom`（或安装其实际来源 feed）即可继续构建；当前仓库无需修改，关闭 issue。

closed 2026-09-05 by coolsnowwolf