coolsnowwolf/lede #12407: R4S 编译失败了

BenjaminX, 2024-08-14:
### 详细叙述

大概率是 leda 更新了 Rockchip 的 SoC 内核

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

R4S

### 详细日志

[image]
make[3]: Entering directory '/workdir/openwrt/package/boot/uboot-envtools'
mkdir -p /workdir/openwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-aarch64_generic_musl/u-boot-2023.01/.pkgdir/uboot-envtools/. /workdir/openwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /workdir/openwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.uboot-envtools_installed
echo "uboot-envtools" >> /workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/uboot-envtools.default.install
make[3]: Leaving directory '/workdir/openwrt/package/boot/uboot-envtools'
time: package/boot/uboot-envtools/compile#0.25#0.18#0.33
make[3]: Entering directory '/workdir/openwrt/package/boot/uboot-rk35xx'
rm -f /workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rockchip-rk3528-evb/uboot-rockchip-2023-07-25-32640b0a/.built
touch /workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rockchip-rk3528-evb/uboot-rockchip-2023-07-25-32640b0a/.built_check
make  -C /workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rockchip-rk3528-evb/uboot-rockchip-2023-07-25-32640b0a CROSS_COMPILE=aarch64-openwrt-linux-musl- DTC="/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.100/scripts/dtc/dtc" HOSTCC="gcc" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-aarch64_generic_musl/host/include -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-aarch64_generic_musl/host/include -std=gnu11" HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib -L/workdir/openwrt/staging_dir/hostpkg/lib -L/workdir/openwrt/staging_dir/target-aarch64_generic_musl/host/lib" LOCALVERSION="-OpenWrt-r6824-2995b9209" STAGING_PREFIX="/workdir/openwrt/staging_dir/host" PKG_CONFIG_PATH="/workdir/openwrt/staging_dir/host/lib/pkgconfig" PKG_CONFIG_LIBDIR="/workdir/openwrt/staging_dir/host/lib/pkgconfig" PKG_CONFIG_EXTRAARGS="--static" V='' IDB_SOC=rk3528 TPL_BIN=/workdir/openwrt/staging_dir/target-aarch64_generic_musl/image/rk3528_ddr_1056MHz_v1.06.bin u-boot.itb idbloader.img
make[4]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rockchip-rk3528-evb/uboot-rockchip-2023-07-25-32640b0a'
  CHK     include/config/uboot.release
  CHK     include/generated/version_autogenerated.h
  CHK     include/generated/timestamp_autogenerated.h
  CHK     include/generated/generic-asm-offsets.h
  CHK     include/generated/asm-offsets.h
start=$(aarch64-openwrt-linux-musl-nm u-boot | grep __rel_dyn_start | cut -f 1 -d ' '); end=$(aarch64-openwrt-linux-musl-nm u-boot | grep __rel_dyn_end | cut -f 1 -d ' '); tools/relocate-rela u-boot-nodtb.bin 0x00200000 $start $end
  COPY    u-boot.dtb
./"arch/arm/mach-rockchip/make_fit_atf.sh" \
arch/arm/dts/rk3528-evb.dtb > u-boot.its
/usr/bin/env: 'python2': No such file or directory
ls: cannot access 'bl31_0x*.bin': No such file or directory
  MKIMAGE u-boot.itb
Error: u-boot.its:46.29-30 syntax error
FATAL ERROR: Unable to parse input tree
./tools/mkimage: Can't read u-boot.itb.tmp: Invalid argument
make[4]: *** [Makefile:1056: u-boot.itb] Error 255
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rockchip-rk3528-evb/uboot-rockchip-2023-07-25-32640b0a'
make[3]: *** [Makefile:77: /workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rockchip-rk3528-evb/uboot-rockchip-2023-07-25-32640b0a/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/boot/uboot-rk35xx'
time: package/boot/uboot-rk35xx/rk3528-evb/compile#1.18#0.60#1.72
    ERROR: package/boot/uboot-rk35xx failed to build (build variant: rk3528-evb).
make[2]: *** [package/Makefile:116: package/boot/uboot-rk35xx/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

whalechoi, 2024-08-14:
/usr/bin/env: 'python2': No such file or directory

python2 装一下

dragon888888888888, 2024-08-15:
package/boot/uboot-rk35xx/Makefile
47 UBOOT_TARGETS := rk3528-evb rk3588-evb
前面添加#
 #UBOOT_TARGETS := rk3528-evb rk3588-evb

BenjaminX, 2024-08-16:
> /usr/bin/env: 'python2': No such file or directory
> 
> python2 装一下

肯定不是装 Python2 的问题

dragon888888888888, 2024-08-16:
> package/boot/uboot-rk35xx/Makefile
> 47行 UBOOT_TARGETS := rk3528-evb rk3588-evb
> 前面添加#
> #UBOOT_TARGETS := rk3528-evb rk3588-evb
进这个选项看看就知道了
[image]

lanruihua026, 2024-08-16:
添加如下命令
sed -i '/^UBOOT_TARGETS := rk3528-evb rk3588-evb/s/^/#/' package/boot/uboot-rk35xx/Makefile

BenjaminX, 2024-08-16:
未做修改，编译通过了。