coolsnowwolf/lede #12410: x64编译失败

ChaingTsung, 2024-08-15:
### 详细叙述

x64编译失败

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x64

### 详细日志

...
mkdir -p /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/build_dir/target-x86_64_musl/linux-firmware-20240709/.pkgdir/wl18xx-firmware/. /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.wl18xx-firmware_installed
echo "wl18xx-firmware" >> /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
make[3]: Leaving directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#2.29#0.58#2.85
make[3]: Entering directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/package/kernel/linux'
mkdir -p /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
Package kmod-usb-audio is missing dependencies for the following libraries:
mc.ko
make[3]: *** [modules/usb.mk:590: /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/bin/targets/x86/64/packages/kmod-usb-audio_6.6.40-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/package/kernel/linux'
time: package/kernel/linux/compile#3.83#0.48#4.29
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt'
make[1]: *** [package/Makefile:110: /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt'
make: *** [/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/include/toplevel.mk:231: world] Error 2

lanruihua026, 2024-08-19:
**Package kmod-usb-audio is missing dependencies for the following libraries:
mc.ko**
尝试取消该插件再尝试编译

coolsnowwolf, 2026-09-05:
日志中的实际失败是 `kmod-usb-audio` 打包时找不到 `mc.ko`。该问题已由提交 `b7dd70a0d70`（`kernel: usb: kmod-usb-audio: Fix dependency`）修复：当前 `package/kernel/linux/modules/usb.mk` 已使用 `DEPENDS+=+kmod-media-controller`，会正确拉入提供 `mc.ko` 的模块包。

原 issue 使用的是 2024 年旧代码和旧内核 6.6.40，当前依赖定义已不同，无需新增修改，关闭 issue。

closed 2026-09-05 by coolsnowwolf