coolsnowwolf/lede #12411: R68s云编译失败

qqsir-dev, 2024-08-15:
### 详细叙述

Github 云编译失败

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

ockchip_armv8_DEVICE_fastrhino_r68s

### 详细日志

make[4]: Entering directory '/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/uboot-rockchip-rk3528-evb/uboot-rockchip-2023-07-25-32640b0a'
  CHK     include/config/uboot.release
  CHK     include/generated/version_autogenerated.h
  CHK     include/generated/timestamp_autogenerated.h
  CHK     include/generated/generic-asm-offsets.h
  CHK     include/generated/asm-offsets.h
start=$(aarch64-openwrt-linux-musl-nm u-boot | grep __rel_dyn_start | cut -f 1 -d ' '); end=$(aarch64-openwrt-linux-musl-nm u-boot | grep __rel_dyn_end | cut -f 1 -d ' '); tools/relocate-rela u-boot-nodtb.bin 0x00200000 $start $end
  COPY    u-boot.dtb
./"arch/arm/mach-rockchip/make_fit_atf.sh" \
arch/arm/dts/rk3528-evb.dtb > u-boot.its
/usr/bin/env: 'python2': No such file or directory
ls: cannot access 'bl31_0x*.bin': No such file or directory
  MKIMAGE u-boot.itb
Error: u-boot.its:46.29-30 syntax error
FATAL ERROR: Unable to parse input tree
./tools/mkimage: Can't read u-boot.itb.tmp: Invalid argument
make[4]: *** [Makefile:1056: u-boot.itb] Error 255
make[4]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/uboot-rockchip-rk3528-evb/uboot-rockchip-2023-07-25-32640b0a'
make[3]: *** [Makefile:77: /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-aarch64_generic_musl/uboot-rockchip-rk3528-evb/uboot-rockchip-2023-07-25-32640b0a/.built] Error 2
make[3]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt/package/boot/uboot-rk35xx'
time: package/boot/uboot-rk35xx/rk3528-evb/compile#1.22#0.58#1.73
    ERROR: package/boot/uboot-rk35xx failed to build (build variant: rk3528-evb).
make[2]: *** [package/Makefile:116: package/boot/uboot-rk35xx/compile] Error 1
make[2]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt'
make[1]: *** [package/Makefile:110: /home/runner/work/AutoLede/AutoLede/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt'
make: *** [/home/runner/work/AutoLede/AutoLede/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

dragon888888888888, 2024-08-15:
需要修改源码
如果云编译无法修改源码，那就没办法了
package/boot/uboot-rk35xx/Makefile
47行 UBOOT_TARGETS := rk3528-evb rk3588-evb
前面添加#
#UBOOT_TARGETS := rk3528-evb rk3588-evb

lanruihua026, 2024-08-16:
> 需要修改源码 如果云编译无法修改源码，那就没办法了 package/boot/uboot-rk35xx/Makefile 47行 UBOOT_TARGETS := rk3528-evb rk3588-evb 前面添加# #UBOOT_TARGETS := rk3528-evb rk3588-evb

云编译脚本添加下述命令
sed -i '/^UBOOT_TARGETS := rk3528-evb rk3588-evb/s/^/#/' package/boot/uboot-rk35xx/Makefile

dragon888888888888, 2024-08-21:
> /usr/bin/env: 'python2': No such file or directory
> 
> 编译环境缺少python2，其实actions是有的，只是是叫python2.7，需要自行链接一下

什么都不缺
被强制多选了
[image]

closed 2024-08-30 by qqsir-dev