coolsnowwolf/lede #12415: 编译失败

lianghdx, 2024-08-17:
### 详细叙述

操作系统分配了32GB，提示内存溢出，需要60GiB内存，是否可以调低bs值？

dd if=/mnt/disk/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz of=/mnt/disk/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz.new **bs=64424509440** conv=sync

**dd: memory exhausted by input buffer of size 64424509440 bytes (60 GiB)**

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x64

### 详细日志

cp /mnt/disk/lede/build_dir/target-x86_64_musl/linux-x86_64/bzImage /mnt/disk/lede/build_dir/target-x86_64_musl/linux-x86_64/generic-kernel.bin
cp /mnt/disk/lede/build_dir/target-x86_64_musl/linux-x86_64/generic-kernel.bin /mnt/disk/lede/bin/targets/x86/64/openwrt-x86-64-generic-kernel.bin
[ -f /mnt/disk/lede/build_dir/target-x86_64_musl/linux-x86_64/generic-kernel.bin -a -f /mnt/disk/lede/build_dir/target-x86_64_musl/linux-x86_64/root.squashfs ]
dd if=/mnt/disk/lede/build_dir/target-x86_64_musl/linux-x86_64/root.squashfs >> /mnt/disk/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz
115017+1 records in
115017+1 records out
58889127 bytes (59 MB, 56 MiB) copied, 1.10456 s, 53.3 MB/s
dd if=/mnt/disk/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz of=/mnt/disk/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz.new **bs=64424509440** conv=sync
**dd: memory exhausted by input buffer of size 64424509440 bytes (60 GiB)**
make[5]: *** [Makefile:164: /mnt/disk/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img.gz] Error 1
make[5]: Leaving directory '/mnt/disk/lede/target/linux/x86/image'
make[4]: *** [Makefile:26: install] Error 2
make[4]: Leaving directory '/mnt/disk/lede/target/linux/x86'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/mnt/disk/lede/target/linux'
time: target/linux/install#137.50#34.12#79.88
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/mnt/disk/lede'
make[1]: *** [target/Makefile:24: /mnt/disk/lede/staging_dir/target-x86_64_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/mnt/disk/lede'
make: *** [/mnt/disk/lede/include/toplevel.mk:231: world] Error 2
[image]

hcym, 2024-08-18:
虚拟机不清楚，实体的一次编译至少40，如果插件特别多，48G都不够，

lianghdx, 2024-08-18:
问题是解决了，与kernel分区大小和root分区大小有关，调小分区大小问题解决。但是，现在的硬盘空间都比较大，这里分区设置大一点应该也有不少人设置，如果能调小bs，用count的话，应该能减轻内存的压力。

coolsnowwolf, 2024-08-19:
dd 临时生成的管道文件就是要占用对应的内存的

hanwei996, 2024-08-26:
> 问题是解决了，与kernel分区大小和root分区大小有关，调小分区大小问题解决。但是，现在的硬盘空间都比较大，这里分区设置大一点应该也有不少人设置，如果能调小bs，用count的话，应该能减轻内存的压力。

咋搞, 也遇到这问题了

hanwei996, 2024-08-26:
看不出是那个目录下的  Makefile  执行的 dd 命令, 如果知道内存小些都可以解决

lianghdx, 2024-08-30:
> 看不出是那个目录下的 Makefile 执行的 dd 命令, 如果知道内存小些都可以解决

方法一：
调低kernel内核分区大小和root分区大小，恢复到默认的大小就行了，我后来调整的是kernel 16MB，root 2048MB。
[image]

方法二：挂载交换分区
虚拟机挂载一个新硬盘，稍微大一点，我当时用的是100GB，然后以root用户执行以下命令
fdisk /dev/sdc（新挂载磁盘所在路径,，创建一个swap分区）
mkswap /dev/sdc1（此处为交换分区路径，格式化交换分区）
swapon /dev/sdc1 （此处为交换分区路径，挂载交换分区，每次使用都需要挂载，或者在/etc/fstab中写入挂载信息，这样每次开机系统会自动挂载，新手不建议操作/etc/fstab，编辑错误将导致系统无法开机）
free -h（检查交换分区是否挂载成功）

closed 2024-09-30 by lianghdx