coolsnowwolf/lede #12416: h68k无法识别sata硬盘

jiawen94, 2024-08-18:
### 详细叙述

h68k使用原厂的usb-c转sata的转接板，无法识别到硬盘。我记得去年6月份编译的固件版本是可以正常识别sata转接板上的硬盘的。
dmesg查看日志未看到sata关键字的日志信息
如需内核日志等信息，我再还原环境提供

已知信息
```
root@OpenWrt:~# uname -a
Linux OpenWrt 6.1.100 #0 SMP PREEMPT Mon Aug 12 08:52:24 2024 aarch64 GNU/Linux
```

固件版本 OpenWrt R24.8.1 / LuCI Master (git-24.182.36151-66d29de)

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

h68k

### 详细日志

无

aiamadeus, 2024-08-19:
https://github.com/coolsnowwolf/lede/commit/54ad1c5009b07dca48eeaf7650345b41e1abcf66 6.1 要选上 kmod-ata-ahci-dwc

jiawen94, 2024-08-19:
> [54ad1c5](https://github.com/coolsnowwolf/lede/commit/54ad1c5009b07dca48eeaf7650345b41e1abcf66) 6.1 要选上 kmod-ata-ahci-dwc

你好，确认一下，只要选上`CONFIG_PACKAGE_kmod-ata-ahci-platform`，下列清单中其他的不需要是吗？
```
# CONFIG_PACKAGE_kmod-ata-ahci-dwc is not set
# CONFIG_PACKAGE_kmod-ata-ahci-platform is not set
# CONFIG_PACKAGE_kmod-ata-artop is not set
# CONFIG_PACKAGE_kmod-ata-core is not set
# CONFIG_PACKAGE_kmod-ata-nvidia-sata is not set
# CONFIG_PACKAGE_kmod-ata-pdc202xx-old is not set
# CONFIG_PACKAGE_kmod-ata-piix is not set
# CONFIG_PACKAGE_kmod-ata-sil is not set
# CONFIG_PACKAGE_kmod-ata-sil24 is not set
# CONFIG_PACKAGE_kmod-ata-via-sata is not set
```

jiawen94, 2024-08-22:
添加 CONFIG_PACKAGE_kmod-ata-ahci-dwc 参数后，SATA可以正常识别了

closed 2024-08-22 by jiawen94