coolsnowwolf/lede #12417: R5S编译失败，哪位指导以下

raymondseek, 2024-08-19:
### 详细叙述

x86编译全部正常，RK如下错误日志，请帮助定位问题

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

R5S

### 详细日志

arch/arm/dts/rk3528-evb.dtb > u-boot.its
/usr/bin/env: 'python2': No such file or directory
ls: cannot access 'bl31_0x*.bin': No such file or directory
  MKIMAGE u-boot.itb
Error: u-boot.its:46.29-30 syntax error
FATAL ERROR: Unable to parse input tree
./tools/mkimage: Can't read u-boot.itb.tmp: Invalid argument
make[4]: *** [Makefile:1056: u-boot.itb] Error 255
make[4]: Leaving directory '/home/raymond/lede/build_dir/target-aarch64_generic_musl/uboot-rockchip-rk3528-evb/uboot-rockchip-2023-07-25-32640b0a'
make[3]: *** [Makefile:77: /home/raymond/lede/build_dir/target-aarch64_generic_musl/uboot-rockchip-rk3528-evb/uboot-rockchip-2023-07-25-32640b0a/.built] Error 2
make[3]: Leaving directory '/home/raymond/lede/package/boot/uboot-rk35xx'
time: package/boot/uboot-rk35xx/rk3528-evb/compile#0.95#0.69#1.63
    ERROR: package/boot/uboot-rk35xx failed to build (build variant: rk3528-evb).
make[2]: *** [package/Makefile:116: package/boot/uboot-rk35xx/compile] Error 1
make[2]: Leaving directory '/home/raymond/lede'
make[1]: *** [package/Makefile:110: /home/raymond/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/raymond/lede'
make: *** [/home/raymond/lede/include/toplevel.mk:231：world] 错误 2

caiwx86, 2024-08-19:
sed -i '/^UBOOT_TARGETS := rk3528-evb rk3588-evb/s/^/#/' package/boot/uboot-rk35xx/Makefile

ghmole, 2024-08-20:
same issue

science2468, 2024-08-20:
看错误是依赖的问题。你把错误放到通义ai问问。

raymondseek, 2024-08-20:
> sed -i '/^UBOOT_TARGETS := rk3528-evb rk3588-evb/s/^/#/' package/boot/uboot-rk35xx/Makefile
好神奇的一行命令，问题解决了，正常编译。请问这个命令解决的是什么问题》？谢谢

science2468, 2024-08-20:
> sed -i '/^UBOOT_TARGETS := rk3528-evb rk3588-evb/s/^/#/' package/boot/uboot-rk35xx/Makefile
作用如下：
#UBOOT_TARGETS := rk3528-evb rk3588-evb
加了个”#“将这行注释掉了。

xyz201704, 2024-08-20:
一样的问题，用前面给出的sed -i指令可以解决

closed 2024-11-24 by aiamadeus