coolsnowwolf/lede #12418: Naive编译错误，crtbeginS.o: No such file or directory

grassrecluse, 2024-08-20:
### 详细叙述

[4009/4009] LINK ./naive
FAILED: naive 
"python3" "../../build/toolchain/gcc_link_wrapper.py" --output="./naive" -- ../../third_party/llvm-build/Release+Asserts/bin/clang++ -fuse-ld=lld -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,--icf=all -Wl,--color-diagnostics -Wl,-mllvm,-instcombine-lower-dbg-declare=0 -Wl,-mllvm,-split-threshold-for-reg-with-hint=0 -Wl,--thinlto-cache-dir=thinlto-cache -Wl,--thinlto-cache-policy=cache_size=10\%:cache_size_bytes=40g:cache_size_files=100000 -flto=thin -Wl,--thinlto-jobs=all -Wl,-mllvm,-import-instr-limit=30 -Wl,-mllvm,-disable-auto-upgrade-debug-info -fwhole-program-vtables -mcpu=cortex-a9 -Wl,--undefined-version --target=arm-openwrt-linux-musleabi -mfloat-abi=hard -no-canonical-prefixes -Wl,-O2 -Wl,--gc-sections -Wl,-z,defs -Wl,--as-needed -nostdlib++ --sysroot=../../../../../../staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.3.0_musl_eabi -Wl,--lto-O2 -Wl,-mllvm,-enable-ext-tsp-block-placement=1 -rdynamic -pie -Wl,--disable-new-dtags -o "./naive" -Wl,--start-group @"./naive.rsp"  -Wl,--end-group  -ldl -lpthread -lrt -lresolv -latomic 
ld.lld: error: cannot open crtbeginS.o: No such file or directory
ld.lld: error: unable to find library -lgcc
ld.lld: error: unable to find library -lgcc
ld.lld: error: cannot open crtendS.o: No such file or directory
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

WRT32X

### 详细日志

[4009/4009] LINK ./naive
FAILED: naive 
"python3" "../../build/toolchain/gcc_link_wrapper.py" --output="./naive" -- ../../third_party/llvm-build/Release+Asserts/bin/clang++ -fuse-ld=lld -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,--icf=all -Wl,--color-diagnostics -Wl,-mllvm,-instcombine-lower-dbg-declare=0 -Wl,-mllvm,-split-threshold-for-reg-with-hint=0 -Wl,--thinlto-cache-dir=thinlto-cache -Wl,--thinlto-cache-policy=cache_size=10\%:cache_size_bytes=40g:cache_size_files=100000 -flto=thin -Wl,--thinlto-jobs=all -Wl,-mllvm,-import-instr-limit=30 -Wl,-mllvm,-disable-auto-upgrade-debug-info -fwhole-program-vtables -mcpu=cortex-a9 -Wl,--undefined-version --target=arm-openwrt-linux-musleabi -mfloat-abi=hard -no-canonical-prefixes -Wl,-O2 -Wl,--gc-sections -Wl,-z,defs -Wl,--as-needed -nostdlib++ --sysroot=../../../../../../staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.3.0_musl_eabi -Wl,--lto-O2 -Wl,-mllvm,-enable-ext-tsp-block-placement=1 -rdynamic -pie -Wl,--disable-new-dtags -o "./naive" -Wl,--start-group @"./naive.rsp"  -Wl,--end-group  -ldl -lpthread -lrt -lresolv -latomic 
ld.lld: error: cannot open crtbeginS.o: No such file or directory
ld.lld: error: unable to find library -lgcc
ld.lld: error: unable to find library -lgcc
ld.lld: error: cannot open crtendS.o: No such file or directory
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.

yanxin152133, 2024-08-20:
https://github.com/coolsnowwolf/lede/issues/6268

coolsnowwolf, 2026-09-05:
该日志对应旧版 NaiveProxy 构建：链接器使用 GCC 11.3.0 目标目录，却找不到 `crtbeginS.o`/`crtendS.o` 和 `libgcc`。当前 `feeds/helloworld/naiveproxy` 已通过 `target_sysroot` 指向 OpenWrt 工具链，并在 ARM `muslgnueabi` 与 LLVM 使用的 `musleabi` 目录名之间建立兼容链接；当前 NaiveProxy 版本和构建脚本也已更新。

issue 没有提供当前提交或复现配置，现有错误无法在当前代码上重现，也不应通过硬编码某个旧 GCC 版本路径修复。请确保先完成目标工具链构建后再编译 NaiveProxy。本 issue 关闭；若当前版本仍失败，请附新的完整日志。

closed 2026-09-05 by coolsnowwolf