coolsnowwolf/lede #12421: wireguard 速度慢，找了很久都没找到问题，等待大佬诊脉

jsdfhasuh, 2024-08-20:
### 详细叙述

wireguard的速度很慢，我的wireguard是连接国内两台机器做异地组网的。正常走域名能有几m/s

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

am6b-plus

### 详细日志

编译正常

jsdfhasuh, 2024-08-21:
[image]
从192.168.1.1用iperf3 测到192.168.100.1
[image]
从192.168.100.1到192.168.1.1是正常的，单边不正常

ZqinKing, 2024-08-26:
不是MTU设的不对，就是运营商的QOS。wireguard的MTU修改为1280，端口修改为5060尝试一下。

ZqinKing, 2024-09-02:
我在N1上遇到了相同的问题。在使用f大的内核打包时出现。调试了各种参数后无法解决。最后通过降级内核至5.15解决。6.1和6.6的内核均出现了相同现象。

jsdfhasuh, 2024-09-02:
很有可能，我晚点试试，我现在正常的OpenWrt上面使用的就是5.15内核

jsdfhasuh, 2024-09-02:
> 我在N1上遇到了相同的问题。在使用f大的内核打包时出现。调试了各种参数后无法解决。最后通过降级内核至5.15解决。6.1和6.6的内核均出现了相同现象。

请问用的是5.15的那个版本？

ZqinKing, 2024-09-02:
> > 我在N1上遇到了相同的问题。在使用f大的内核打包时出现。调试了各种参数后无法解决。最后通过降级内核至5.15解决。6.1和6.6的内核均出现了相同现象。
> 
> 请问用的是5.15的那个版本？

最新的

jsdfhasuh, 2024-09-11:
> > > 我在N1上遇到了相同的问题。在使用f大的内核打包时出现。调试了各种参数后无法解决。最后通过降级内核至5.15解决。6.1和6.6的内核均出现了相同现象。
> > 
> > 请问用的是5.15的那个版本？
> 
> 最新的

用最新的6.6内核就可以解决，的确是内核问题

closed 2024-09-11 by jsdfhasuh