coolsnowwolf/lede #12423: Xiaomi Redmi Router AX6000自动重启

smithe-ma, 2024-08-21:
### 详细叙述

每24小时内会莫名其妙重启

主机名	OpenWrt
主机型号	Xiaomi Redmi Router AX6000 (CpuMark : 27777.777778 Scores)
架构	ARMv8 Processor x 4 (45.2°C)
固件版本	OpenWrt R24.7.7 / LuCI Master (git-24.182.36151-66d29de)
内核版本	6.1.97 

开启如下设置：
软件流量分载
基于软件的 Routing/NAT 分载
硬件流量分载
需要硬件 NAT 支持，目前 mt762x 已实现
MTK 无线硬件加速
需要硬件支持，目前 Filogic 8x0 已实现
BBR 拥塞控制算法
使用 BBR 拥塞控制算法可以有效提升 TCP 网络性能 

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

Xiaomi Redmi Router AX6000

### 详细日志

内核日志

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.1.97 (admin@iZj6c7khbkoi3ts19wnkjrZ) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r6769-665cedd25) 11.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Sat Jul 13 15:02:08 2024
[    0.000000] Machine model: Xiaomi Redmi Router AX6000
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
[    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
[    0.000000]   node   0: [mem 0x0000000043030000-0x000000004fbfffff]
[    0.000000]   node   0: [mem 0x000000004fc00000-0x000000004ffbffff]
[    0.000000]   node   0: [mem 0x000000004ffc0000-0x000000005fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 18 pages/cpu s33768 r8192 d31768 u73728
[    0.000000] pcpu-alloc: s33768 r8192 d31768 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] Kernel command line: console=ttyS0,115200n1 loglevel=8 firmware=1 uart_en=1
[    0.000000] Unknown kernel command line parameters "firmware=1 uart_en=1", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 497108K/524288K available (8448K kernel code, 910K rwdata, 2432K rodata, 448K init, 292K bss, 27180K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[... 228 lines cut ...]
[   29.729355] br-lan: port 2(lan3) entered disabled state
[   29.736258] device lan3 entered promiscuous mode
[   29.743930] mt7530-mdio mdio-bus:1f lan4: configuring for phy/gmii link mode
[   29.753589] br-lan: port 3(lan4) entered blocking state
[   29.758818] br-lan: port 3(lan4) entered disabled state
[   29.765812] device lan4 entered promiscuous mode
[   29.774228] mt7530-mdio mdio-bus:1f wan: configuring for phy/gmii link mode
[   29.788714] mt7530-mdio mdio-bus:1f wan: Link is Up - 100Mbps/Full - flow control off
[   29.796595] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
[   32.350545] br-lan: port 4(wlan0) entered blocking state
[   32.355889] br-lan: port 4(wlan0) entered disabled state
[   32.361413] device wlan0 entered promiscuous mode
[   35.306951] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   35.313458] br-lan: port 4(wlan0) entered blocking state
[   35.318794] br-lan: port 4(wlan0) entered forwarding state
[   35.324449] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   35.454345] br-lan: port 5(wlan1) entered blocking state
[   35.459734] br-lan: port 5(wlan1) entered disabled state
[   35.465286] device wlan1 entered promiscuous mode
[   35.470125] br-lan: port 5(wlan1) entered blocking state
[   35.475429] br-lan: port 5(wlan1) entered forwarding state
[   36.325648] br-lan: port 5(wlan1) entered disabled state
[   36.624145] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   36.630640] br-lan: port 5(wlan1) entered blocking state
[   36.635964] br-lan: port 5(wlan1) entered forwarding state

Turing-China, 2024-08-21:
我的重启不会;但是编译好了手机和电脑都找不到WIFI信号

smithe-ma, 2024-08-21:
@aiamadeus 怎么升级6.1.100，重新编译最新的版本吗？

aiamadeus, 2024-08-21:
没机子没启动日志修不了，不关这两个 commit 事
又不是我强行更新 6.x 内核的，别一直 at 我。。

Turing-China, 2024-08-22:
> @Turing-China 2.4和5G一切正常。网络都是好的。就是莫名奇妙重启。我装的东西很少。你也是这个内核和版本吗

@smithe-ma 你重新编译最新内核就是最新的版本  另外我5.5 6.1 6.6内核都试过 全部都是能看到WIFI配置 但是搜索不到WIFI信号;你是云编译的吗?能不能把config 文件发一下；感谢

closed 2024-10-21 by smithe-ma