coolsnowwolf/lede #12426: WCN6855 / NFA765 在PVE虚拟机中直通的问题

Kisoul, 2024-08-21:
### 详细叙述

PVE设置NFA765直通给openwrt。其他 WiFi 卡工作正常，但765网卡无法启动也无法设置参数。如图：
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

qcnfa765

### 详细日志

Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.370543] acpiphp: Slot [29-1] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.371111] acpiphp: Slot [30] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.371667] acpiphp: Slot [31] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.372429] pci 0000:00:05.0: PCI bridge to [bus 01]
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.373072] pci 0000:00:05.0:   bridge window [io  0xe000-0xefff]
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.373836] pci 0000:00:05.0:   bridge window [mem 0xfe200000-0xfe3fffff]
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.374690] pci 0000:00:05.0:   bridge window [mem 0xfd400000-0xfd5fffff 64bit pref]
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.376308] pci_bus 0000:02: extended config space not accessible
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.377080] acpiphp: Slot [27-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.377654] acpiphp: Slot [0-1] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.378220] acpiphp: Slot [1-1] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.378774] acpiphp: Slot [2-1] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.379335] acpiphp: Slot [3-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.379896] acpiphp: Slot [4-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.380462] acpiphp: Slot [5-1] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.381015] acpiphp: Slot [6-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.381574] acpiphp: Slot [7-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.382133] acpiphp: Slot [8-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.382690] acpiphp: Slot [9-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.383247] acpiphp: Slot [10-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.383808] acpiphp: Slot [11-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.384381] acpiphp: Slot [12-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.384945] acpiphp: Slot [13-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.385502] acpiphp: Slot [14-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.386072] acpiphp: Slot [15-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.386643] acpiphp: Slot [16-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.387210] acpiphp: Slot [17-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.387770] acpiphp: Slot [18-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.388343] acpiphp: Slot [19-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.388908] acpiphp: Slot [20-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.389474] acpiphp: Slot [21-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.390037] acpiphp: Slot [22-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.390608] acpiphp: Slot [23-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.391179] acpiphp: Slot [24-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.391740] acpiphp: Slot [25-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.392310] acpiphp: Slot [26-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.392876] acpiphp: Slot [28-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.393450] acpiphp: Slot [29-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.395502] acpiphp: Slot [30-1] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.396070] acpiphp: Slot [31-1] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.396869] pci 0000:02:1b.0: [1b36:000d] type 00 class 0x0c0330
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.398277] pci 0000:02:1b.0: reg 0x10: [mem 0xfe000000-0xfe003fff 64bit]
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.402567] pci 0000:00:1e.0: PCI bridge to [bus 02]
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.403226] pci 0000:00:1e.0:   bridge window [io  0xd000-0xdfff]
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.403993] pci 0000:00:1e.0:   bridge window [mem 0xfe000000-0xfe1fffff]
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.404840] pci 0000:00:1e.0:   bridge window [mem 0xfd200000-0xfd3fffff 64bit pref]
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.406291] pci_bus 0000:03: extended config space not accessible
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.407137] acpiphp: Slot [13-3] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.407733] acpiphp: Slot [14-3] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.408326] acpiphp: Slot [0-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.409650] acpiphp: Slot [1-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.410237] acpiphp: Slot [2-2] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.410801] acpiphp: Slot [3-3] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.411365] acpiphp: Slot [4-3] registered
Wed Aug 21 23:26:08 2024 kern.info kernel: [    0.411924] acpiphp: Slot [5-2] registered
[... 513 lines cut ...]
Wed Aug 21 23:26:15 2024 daemon.info aria2: Instance "main" disabled.
Wed Aug 21 23:26:16 2024 daemon.notice procd: /etc/rc.d/S99nfsd: exportfs: /etc/exports [1]: Neither 'subtree_check' or 'no_subtree_check' specified for export "*:/mnt".
Wed Aug 21 23:26:16 2024 daemon.notice procd: /etc/rc.d/S99nfsd:   Assuming default behaviour ('no_subtree_check').
Wed Aug 21 23:26:16 2024 daemon.notice procd: /etc/rc.d/S99nfsd:   NOTE: this default has changed since nfs-utils version 1.0.x
Wed Aug 21 23:26:16 2024 daemon.notice procd: /etc/rc.d/S99nfsd:
Wed Aug 21 23:26:16 2024 daemon.notice procd: /etc/rc.d/S99nfsd: exportfs: /mnt does not support NFS export
Wed Aug 21 23:26:16 2024 daemon.err nmbd[5802]: [2024/08/21 23:26:16.438159,  0] ../../lib/util/become_daemon.c:135(daemon_ready)
Wed Aug 21 23:26:16 2024 daemon.err nmbd[5802]:   daemon_ready: daemon 'nmbd' finished starting up and ready to serve connections
Wed Aug 21 23:26:16 2024 daemon.err smbd[5801]: [2024/08/21 23:26:16.532986,  0] ../../lib/util/become_daemon.c:135(daemon_ready)
Wed Aug 21 23:26:16 2024 daemon.err smbd[5801]:   daemon_ready: daemon 'smbd' finished starting up and ready to serve connections
Wed Aug 21 23:26:17 2024 kern.warn kernel: [   18.523038] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
Wed Aug 21 23:26:17 2024 kern.info kernel: [   18.523939] NFSD: Using legacy client tracking operations.
Wed Aug 21 23:26:17 2024 kern.info kernel: [   18.524586] NFSD: starting 10-second grace period (net f0000000)
Wed Aug 21 23:26:17 2024 daemon.notice procd: /etc/rc.d/S99openclash: OpenClash Restart...
Wed Aug 21 23:26:17 2024 daemon.notice rpc.statd[6734]: Version 2.6.2 starting
Wed Aug 21 23:26:17 2024 daemon.warn rpc.statd[6734]: Flags: No-Daemon TI-RPC
Wed Aug 21 23:26:17 2024 daemon.notice rpc.mountd[6735]: Version 2.6.2 starting
Wed Aug 21 23:26:17 2024 daemon.notice rpc.statd[6734]: Initializing NSM state
Wed Aug 21 23:26:17 2024 daemon.warn rpc.statd[6734]: Running as root.  chown /var/lib/nfs to choose different user
Wed Aug 21 23:26:20 2024 kern.warn kernel: [   21.091401] ath11k_pci 0000:00:10.0: failed to wait wlan mode request (mode 4): -110
Wed Aug 21 23:26:20 2024 kern.warn kernel: [   21.092264] ath11k_pci 0000:00:10.0: qmi failed to send wlan mode off: -110
Wed Aug 21 23:26:21 2024 daemon.notice procd: /etc/rc.d/S99openclash: OpenClash Already Stop!
Wed Aug 21 23:26:21 2024 cron.err crond[8182]: crond (busybox 1.36.0) started, log level 5
Wed Aug 21 23:26:21 2024 daemon.notice procd: /etc/rc.d/S99sing-box: sing-box is started!
Wed Aug 21 23:26:21 2024 cron.err crond[8532]: crond (busybox 1.36.0) started, log level 5

Kisoul, 2024-08-28:
> 虚拟机很难直通这类型的网卡

是的，esxi我也是了一样的问题，目前只能用7922，但是功率有点问题

rosaceous, 2024-09-27:
邮件列表上有关于ath11k驱动的讨论

https://lore.kernel.org/all/26aba780-d175-46ff-9943-c36625e7757a@gmail.com/T/

简而言之，高通的网卡固件会向驱动程序请求设备真实的中断地址，而虚拟机中网卡驱动能传递的中断地址是通过IOMMU转译过的，所以固件使用错误的中断地址是不能正常工作的。

Kisoul, 2024-09-28:
> 邮件列表上有关于ath11k驱动的讨论
> 
> https://lore.kernel.org/all/26aba780-d175-46ff-9943-c36625e7757a@gmail.com/T/
> 
> 简而言之，高通的网卡固件会向驱动程序请求设备真实的中断地址，而虚拟机中网卡驱动能传递的中断地址是通过IOMMU转译过的，所以固件使用错误的中断地址是不能正常工作的。

谢谢解答，虽然原文我看不明白，那是不是说暂时没有办法解决？除非高通或者pve做出改变？

rosaceous, 2024-09-28:
PVE只是个linux发行版，他们应该没有能力/动力去解决这个问题。邮件列表里内核开发者认为这个网卡驱动想干的事很疯狂，这个世界上没操作系统提供这个驱动要的功能，而且我认为高通不大可能为这种非常罕见的应用场景修改它们的固件。

作为短期方案，根据邮件列表中的讨论内容，你可以在关闭IOMMU并启动后，在宿主机确定真实的设备中断地址，然后改内核中的驱动代码，写死中断地址。但是，重启几次可能就不管用了，因为中断地址每次启动可能不一样。。。

Kisoul, 2024-09-30:
> PVE只是个linux发行版，他们应该没有能力/动力去解决这个问题。邮件列表里内核开发者认为这个网卡驱动想干的事很疯狂，这个世界上没操作系统提供这个驱动要的功能（所以他们也不会提供这项功能），而且我认为高通不大可能为这种非常罕见的应用场景修改它们的固件，固件这种东西也只有他们自己能修改。
> 
> 作为短期方案，根据邮件列表中的讨论内容，你可以在宿主机上确定真实的设备中断地址，然后改客户机内核中的驱动代码，让驱动使用真实的中断地址。但是，重启几次可能就不管用了，因为中断地址每次启动可能不一样。。。

感谢解答，这个已经超出我能力范围了，与其研究解决不如换个卡。此问题可以结束了 。

closed 2024-09-30 by Kisoul