coolsnowwolf/lede #12429: 试过几十次了，老是卡在这个地方

Ikaros520544, 2024-08-22:
### 详细叙述

系统：ubuntu24.04LTS
CPU：I7-7700K
内存：24GB

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86_64

### 详细日志

SHELL= flock /home/ikaros/桌面/lede/tmp/.root-copy.flock -c 'cp -fpR /home/ikaros/桌面/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kmod-drm-kms-helper/. /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-drm-kms-helper_installed
echo "kmod-drm-kms-helper" >> /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
mkdir -p /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/ikaros/桌面/lede/tmp/.root-copy.flock -c 'cp -fpR /home/ikaros/桌面/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kmod-drm-suballoc-helper/. /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-drm-suballoc-helper_installed
echo "kmod-drm-suballoc-helper" >> /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
mkdir -p /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/ikaros/桌面/lede/tmp/.root-copy.flock -c 'cp -fpR /home/ikaros/桌面/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kmod-drm-amdgpu/. /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-drm-amdgpu_installed
echo "kmod-drm-amdgpu" >> /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
mkdir -p /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/ikaros/桌面/lede/tmp/.root-copy.flock -c 'cp -fpR /home/ikaros/桌面/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kmod-drm-i915/. /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-drm-i915_installed
echo "kmod-drm-i915" >> /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
mkdir -p /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/ikaros/桌面/lede/tmp/.root-copy.flock -c 'cp -fpR /home/ikaros/桌面/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kmod-amd-xgbe/. /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-amd-xgbe_installed
echo "kmod-amd-xgbe" >> /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/pkginfo/linux.default.install
make[3]: Leaving directory '/home/ikaros/桌面/lede/package/kernel/linux'
time: package/kernel/linux/compile#4.38#0.96#5.49
make[3]: Entering directory '/home/ikaros/桌面/lede/package/system/ubox'
mkdir -p /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/ikaros/桌面/lede/tmp/.root-copy.flock -c 'cp -fpR /home/ikaros/桌面/lede/build_dir/target-x86_64_musl/ubox-2022-08-13-4c7b720b/.pkgdir/ubox/. /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.ubox_installed
echo "ubox" >> /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/pkginfo/ubox.default.install
mkdir -p /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/ikaros/桌面/lede/tmp/.root-copy.flock -c 'cp -fpR /home/ikaros/桌面/lede/build_dir/target-x86_64_musl/ubox-2022-08-13-4c7b720b/.pkgdir/getrandom/. /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.getrandom_installed
echo "getrandom" >> /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/pkginfo/ubox.default.install
mkdir -p /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/ikaros/桌面/lede/tmp/.root-copy.flock -c 'cp -fpR /home/ikaros/桌面/lede/build_dir/target-x86_64_musl/ubox-2022-08-13-4c7b720b/.pkgdir/logd/. /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.logd_installed
echo "logd" >> /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/pkginfo/ubox.default.install
make[3]: Leaving directory '/home/ikaros/桌面/lede/package/system/ubox'
time: package/system/ubox/compile#0.22#0.10#0.31
make[3]: Entering directory '/home/ikaros/桌面/lede/package/libs/ncurses'
make[3]: Leaving directory '/home/ikaros/桌面/lede/package/libs/ncurses'
time: package/libs/ncurses/host-compile#0.09#0.09#0.17
make[3]: Entering directory '/home/ikaros/桌面/lede/package/libs/zlib'
mkdir -p /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/ikaros/桌面/lede/tmp/.root-copy.flock -c 'cp -fpR /home/ikaros/桌面/lede/build_dir/target-x86_64_musl/zlib-1.2.13/.pkgdir/zlib/. /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.zlib_installed
echo "zlib" >> /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/pkginfo/zlib.default.install
make[3]: Leaving directory '/home/ikaros/桌面/lede/package/libs/zlib'
time: package/libs/zlib/compile#0.19#0.10#0.27
make[3]: Entering directory '/home/ikaros/桌面/lede/package/libs/ncurses'
mkdir -p /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/ikaros/桌面/lede/tmp/.root-copy.flock -c 'cp -fpR /home/ikaros/桌面/lede/build_dir/target-x86_64_musl/ncurses-6.4/.pkgdir/terminfo/. /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.terminfo_installed
[... 113 lines cut ...]
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 1209, in communicate
    stdout, stderr = self._communicate(input, endtime, timeout)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 2153, in _communicate
    stdout = self._translate_newlines(stdout,
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 1086, in _translate_newlines
    data = data.decode(encoding, errors)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe6 in position 16: invalid continuation byte

../../../build_dir/target-x86_64_musl/util-linux-2.38.1/meson.build:214:0: ERROR: Unhandled python exception

    This is a Meson bug and should be reported!
make[3]: *** [Makefile:878: /home/ikaros/桌面/lede/build_dir/target-x86_64_musl/util-linux-2.38.1/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 2
make[3]: Leaving directory '/home/ikaros/桌面/lede/package/utils/util-linux'
time: package/utils/util-linux/compile#1.23#0.16#1.42
    ERROR: package/utils/util-linux failed to build.
make[2]: *** [package/Makefile:116: package/utils/util-linux/compile] Error 1
make[2]: Leaving directory '/home/ikaros/桌面/lede'
make[1]: *** [package/Makefile:110: /home/ikaros/桌面/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ikaros/桌面/lede'
make: *** [/home/ikaros/桌面/lede/include/toplevel.mk:231：world] 错误 2
ikaros@Ikaros:~/桌面/lede$

science2468, 2024-08-22:
https://github.com/science2468/immortalwrt-IB/releases/tag/lede-x64-2024-08-22-04-22-19   现成的

rrkai, 2024-08-23:
就不能放在纯字母目录......

xyz201704, 2024-08-26:
目录里不能有中文

closed 2024-11-24 by aiamadeus