coolsnowwolf/lede #12430: 小米R3G路由5gWiFi无法启用

OldCoding, 2024-08-24:
### 详细叙述

使用最新源码编译的固件，在配置好WiFi后只有2.4g正常工作，5g显示设备未激活
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

MI-R3G

### 详细日志

以下是系统运行日志
Tue Aug 20 17:34:15 2024 kern.notice kernel: [    0.000000] Linux version 5.4.278 (runner@fv-az1789-69) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r6827-9541a7af4)) #0 SMP Tue Aug 20 09:34:00 2024
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] MIPS: machine is Xiaomi Mi Router 3G
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Tue Aug 20 17:34:15 2024 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Tue Aug 20 17:34:15 2024 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Tue Aug 20 17:34:15 2024 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] Zone ranges:
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000]   HighMem  empty
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] Movable zone start for each node
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] Early memory node ranges
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
Tue Aug 20 17:34:15 2024 kern.debug kernel: [    0.000000] On node 0 totalpages: 65536
Tue Aug 20 17:34:15 2024 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Tue Aug 20 17:34:15 2024 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Tue Aug 20 17:34:15 2024 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s26640 r8192 d22512 u57344
Tue Aug 20 17:34:15 2024 kern.debug kernel: [    0.000000] pcpu-alloc: s26640 r8192 d22512 u57344 alloc=14*4096
Tue Aug 20 17:34:15 2024 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
Tue Aug 20 17:34:15 2024 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] Writing ErrCtl register=00076fd0
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] Readback ErrCtl register=00076fd0
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] Memory: 250084K/262144K available (6200K kernel code, 274K rwdata, 1304K rodata, 1232K init, 235K bss, 12060K reserved, 0K cma-reserved, 0K highmem)
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] NR_IRQS: 256
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] CPU Clock: 880MHz, start overclocking
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] CPU Clock: 1100MHz
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xfdb1960d81, max_idle_ns: 440795240157 ns
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 3475018995 ns
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.000008] sched_clock: 32 bits at 550MHz, resolution 1ns, wraps every 3904515583ns
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.007872] Calibrating delay loop... 732.36 BogoMIPS (lpj=3661824)
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.074071] pid_max: default: 32768 minimum: 301
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.078838] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.086035] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.096571] rcu: Hierarchical SRCU implementation.
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.101988] smp: Bringing up secondary CPUs ...
Tue Aug 20 17:34:15 2024 kern.warn kernel: [    2.021726] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Tue Aug 20 17:34:15 2024 kern.warn kernel: [    2.021736] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Tue Aug 20 17:34:15 2024 kern.warn kernel: [    2.021748] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Tue Aug 20 17:34:15 2024 kern.info kernel: [    2.021844] CPU1 revision is: 0001992f (MIPS 1004Kc)
Tue Aug 20 17:34:15 2024 kern.info kernel: [    0.166789] Synchronize counters for CPU 1: done.
Tue Aug 20 17:34:15 2024 kern.warn kernel: [    2.051012] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Tue Aug 20 17:34:15 2024 kern.warn kernel: [    2.051019] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[... 502 lines cut ...]
Sat Aug 24 10:56:17 2024 daemon.notice netifd: radio1 (7723):  -s <socket>:		Set the unix domain socket to connect to
Sat Aug 24 10:56:17 2024 daemon.notice netifd: radio1 (7723):  -t <timeout>:		Set the timeout (in seconds) for a command to complete
Sat Aug 24 10:56:17 2024 daemon.notice netifd: radio1 (7723):  -S:			Use simplified output (for scripts)
Sat Aug 24 10:56:17 2024 daemon.notice netifd: radio1 (7723):  -v:			More verbose output
Sat Aug 24 10:56:17 2024 daemon.notice netifd: radio1 (7723):  -m <type>:		(for monitor): include a specific message type
Sat Aug 24 10:56:17 2024 daemon.notice netifd: radio1 (7723): 			(can be used more than once)
Sat Aug 24 10:56:17 2024 daemon.notice netifd: radio1 (7723):  -M <r|t>		(for monitor): only capture received or transmitted traffic
Sat Aug 24 10:56:17 2024 daemon.notice netifd: radio1 (7723):
Sat Aug 24 10:56:17 2024 daemon.notice netifd: radio1 (7723): Commands:
Sat Aug 24 10:56:17 2024 daemon.notice netifd: radio1 (7723):  - list [<path>]			List objects
Sat Aug 24 10:56:17 2024 daemon.notice netifd: radio1 (7723):  - call <path> <method> [<message>]	Call an object method
Sat Aug 24 10:56:17 2024 daemon.notice netifd: radio1 (7723):  - subscribe <path> [<path>...]	Subscribe to object(s) notifications
Sat Aug 24 10:56:17 2024 daemon.notice netifd: radio1 (7723):  - listen [<path>...]			Listen for events
Sat Aug 24 10:56:17 2024 daemon.notice netifd: radio1 (7723):  - send <type> [<message>]		Send an event
Sat Aug 24 10:56:17 2024 daemon.notice netifd: radio1 (7723):  - wait_for <object> [<object>...]	Wait for multiple objects to appear on ubus
Sat Aug 24 10:56:17 2024 daemon.notice netifd: radio1 (7723):  - monitor				Monitor ubus traffic
Sat Aug 24 10:56:17 2024 daemon.notice netifd: radio1 (7723):
Sat Aug 24 10:56:17 2024 daemon.notice netifd: radio1 (7723): Device setup failed: HOSTAPD_START_FAILED
Sat Aug 24 10:56:17 2024 daemon.notice netifd: Network device 'wlan0' link is up
Sat Aug 24 10:56:18 2024 daemon.info hostapd: wlan0: STA 50:ec:50:41:4f:c2 IEEE 802.11: authenticated
Sat Aug 24 10:56:18 2024 daemon.info hostapd: wlan0: STA 50:ec:50:41:4f:c2 IEEE 802.11: associated (aid 1)
Sat Aug 24 10:56:18 2024 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 50:ec:50:41:4f:c2 auth_alg=open
Sat Aug 24 10:56:18 2024 daemon.info hostapd: wlan0: STA 50:ec:50:41:4f:c2 RADIUS: starting accounting session 5C4145D895CFB209
Sat Aug 24 10:56:18 2024 daemon.info hostapd: wlan0: STA 50:ec:50:41:4f:c2 WPA: pairwise key handshake completed (RSN)
Sat Aug 24 10:56:18 2024 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 50:ec:50:41:4f:c2

aiamadeus, 2024-08-26:
你 5G wifi 的设置贴一下

OldCoding, 2024-08-26:
> 你 5G wifi 的设置贴一下

只修改了名称、加密方式、密码，剩下的都是默认
[image]
[image]

aiamadeus, 2024-08-26:
你这个是 js luci 吧，换回默认的呢

OldCoding, 2024-08-27:
> 你这个是 js luci 吧，换回默认的呢

默认的luci是正常的，但既然出了23.05分支的luci就修一下呗

Hirro029, 2024-10-23:
是不是共性啊？我的EA8500，也是这个问题。但是原来的Luci是好的，现在转到23.05就这样了。

YEYEv50, 2024-11-21:
> > 你这个是 js luci 吧，换回默认的呢
> 
> 默认的luci是正常的，但既然出了23.05分支的luci就修一下呗

用WinSCP进入路由器/etc/config 手动编辑wireless文件。参考这样
	option channel '44'
	option band '5g'
	option htmode 'VHT80'
	option country 'CN'
	option disabled '0'
	option cell_density '0'
	option mu_beamformer '1'
手动固定信道和频宽，WIFI5大概就这样。改完保存重启就行了。具体看自己机子

YEYEv50, 2024-11-21:
> 是不是共性啊？我的EA8500，也是这个问题。但是原来的Luci是好的，现在转到23.05就这样了。

看我上面的回复，手动配置就行。正常能用。应该是LUCI的问题。

binext, 2024-12-01:
> > 你这个是 js luci 吧，换回默认的呢
> 
> 默认的luci是正常的，但既然出了23.05分支的luci就修一下呗

怎么换回默认的啊？

OldCoding, 2025-02-23:
最新版源码好像修复了，缺失的信道选项已经正常出现了

closed 2025-02-23 by OldCoding