coolsnowwolf/lede #12431: R5S没有与风扇相关的pwm驱动

xyz201704, 2024-08-25:
### 详细叙述

R5S有风扇接口，本来想做风扇控制的，但是没有相关的驱动。
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

友善R5S

### 详细日志

无日志，能正常编译完成。

yanxin152133, 2024-08-27:
淘宝买个温控开关比折腾驱动来的简单 ：）

yanxin152133, 2024-08-27:
https://github.com/stupidloud/nanopi-openwrt/issues/1188

xyz201704, 2024-08-27:
啊？

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年08月27日 09:24 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [coolsnowwolf/lede] R5S没有与风扇相关的pwm驱动 (Issue #12431) |

stupidloud/nanopi-openwrt#1188

QYKing233, 2024-08-27:
> https://github.com/stupidloud/nanopi-openwrt/issues/1188

现在是/sys/class/pwm里没有导出pwm通道，得改DTS，你发的链接只是个shell脚本读取cpu温度设置pwm的占空比和周期的。那属于后话了。

yanxin152133, 2024-08-27:
https://wiki.friendlyelec.com/wiki/index.php/NanoPi_R4S/zh#PWM.E9.A3.8E.E6.89.87.E7.9A.84.E8.B0.83.E8.8A.82
搜索了半天只搜到r4s的

yanxin152133, 2024-08-27:
https://github.com/friendlyarm/kernel-rockchip/commit/f74ac319f02e2d22cdd33227e7f167e4232809f9
DTS

xyz201704, 2024-08-27:
> [friendlyarm/kernel-rockchip@f74ac31](https://github.com/friendlyarm/kernel-rockchip/commit/f74ac319f02e2d22cdd33227e7f167e4232809f9) DTS

没啥用，不过R5S也不需要风扇

xyz201704, 2024-08-27:
[image]
duiz对着修改了下，不知道有没有用

yanxin152133, 2024-08-27:
最简单的办法就是淘宝买个温控开关：）

xyz201704, 2024-08-27:
确实

xyz201704, 2024-08-27:
> [friendlyarm/kernel-rockchip@f74ac31](https://github.com/friendlyarm/kernel-rockchip/commit/f74ac319f02e2d22cdd33227e7f167e4232809f9) DTS

[image]
这么改，编译能过

xyz201704, 2024-08-27:
实际上可以考虑直接把istoreos的驱动搬过来用
https://github.com/istoreos/istoreos/blob/istoreos-22.03/target/linux/rockchip/dts/rk3568/rk3568-nanopi-r5s.dts

xyz201704, 2024-08-27:
> 实际上可以考虑直接把istoreos的驱动搬过来用 https://github.com/istoreos/istoreos/blob/istoreos-22.03/target/linux/rockchip/dts/rk3568/rk3568-nanopi-r5s.dts

> 最简单的办法就是淘宝买个温控开关：）

xyz201704, 2024-08-27:
给后面看到的人：这方法不行，最佳解决方法是温控开关或者从usb引电，让风扇一直转

qingtian110, 2024-09-12:
怎么我的 R5S 有这个东西。。。。

[image]

qingtian110, 2024-09-12:
不过我没有风扇。。。

xyz201704, 2024-09-12:
啊？

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年09月12日 12:00 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [coolsnowwolf/lede] R5S没有与风扇相关的pwm驱动 (Issue #12431) |

怎么我的 R5S 有这个东西。。。。

image.png (view on web)

xyz201704, 2024-09-12:
懵逼了

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年09月12日 12:00 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [coolsnowwolf/lede] R5S没有与风扇相关的pwm驱动 (Issue #12431) |

怎么我的 R5S 有这个东西。。。。

image.png (view on web)

xyz201704, 2024-09-12:
不过你好像是原版，没有LEDE标识

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年09月12日 12:00 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [coolsnowwolf/lede] R5S没有与风扇相关的pwm驱动 (Issue #12431) |

怎么我的 R5S 有这个东西。。。。

image.png (view on web)

qingtian110, 2024-09-12:
是的，基于原生，但不是纯原生的，内核  6.6 的，原生对应的内核版本应该是 5.15

xyz201704, 2024-09-12:
我有风扇，我是从USB接口引电的

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年09月12日 12:01 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [coolsnowwolf/lede] R5S没有与风扇相关的pwm驱动 (Issue #12431) |

不过我没有风扇。。。

xyz201704, 2024-09-12:
R5S实际上可以不用风扇

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年09月12日 12:01 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [coolsnowwolf/lede] R5S没有与风扇相关的pwm驱动 (Issue #12431) |

不过我没有风扇。。。

qingtian110, 2024-09-12:
就回到这个 issues 内容，你可以参考下 https://github.com/sbwml/r4s_build_script 他的配置，我是用的这里固件的，默认有 /sys/class/pwm 的路径

xyz201704, 2024-09-12:
内核6.6？我5.15内核的，太新了没啥插件。

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年09月12日 12:03 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [coolsnowwolf/lede] R5S没有与风扇相关的pwm驱动 (Issue #12431) |

是的，基于原生，但不是纯原生的，内核 6.6 的，原生对应的内核版本应该是 5.15

qingtian110, 2024-09-12:
插件 跟 内核版本    理论上两者没有什么直接关联。5.15 能用的插件    内核  6.6 也能

xyz201704, 2024-09-12:
没试过。现在最新的稳定版本是23.05.4，对应内核是5.15.162。

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年09月12日 12:08 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [coolsnowwolf/lede] R5S没有与风扇相关的pwm驱动 (Issue #12431) |

插件 跟 内核版本 理论上两者没有什么直接关联。5.15 能用的插件 内核 6.6 也能

xyz201704, 2024-09-12:
这我就不懂了。没研究过。
我要用的功能很少，就smartdns和mwan3，还有一些统计功能。

| |
xyz201704
|
|
***@***.***
|
---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年09月12日 12:10 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [coolsnowwolf/lede] R5S没有与风扇相关的pwm驱动 (Issue #12431) |

相反，op 能跑的插件，你目前的固件 未必也能跑。因为op 的luci 支持 js 和 lua，也同时能兼容 fw3 与 fw4 的插件

qingtian110, 2024-11-27:
有风扇了

[image]

WilliamHuangcc, 2025-06-20:
> 有风扇了
> 
> [image]

你好，这是哪个固件的？

coolsnowwolf, 2026-09-05:
已对照当前 LEDE 与上游 OpenWrt 的 NanoPi R5S DTS：板级设备树仅定义 GPIO LED、按键、网口等资源，没有连接到风扇接口的 PWM 控制器/引脚节点；内核也没有可据此自动导出的 `/sys/class/pwm` 通道。现有讨论已验证通过外接温控开关或 USB 供电可使用风扇，但这不属于软件驱动缺失。

在没有 FriendlyElec 官方风扇接口原理图及 PWM 引脚复用信息前，新增 DTS 节点属于猜测，可能影响其他复用功能。因此不引入补丁，关闭 issue。

closed 2026-09-05 by coolsnowwolf