coolsnowwolf/lede #12432: 6088编译客户端电脑无法获得IP

ramondelee, 2024-08-25:
### 详细叙述

编译是成功的。之前没有改默认IP192.168.1.1 ，成功刷写，才开始断了好几次客户电脑连接；后面改默认IP为192.168.12.1重新编译、改回默认IP192.168.1.1编译，刷写后客户电脑均无法获得IP。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

TP XDR 6088

### 详细日志

无

yanxin152133, 2024-08-27:
/etc/config/network  发出来

ramondelee, 2024-08-27:
> /etc/config/network 发出来

没有更改过源码，看起来是在不停重启网络，路由的网络接口灯一直一起亮又黑一会又亮一会，SSH上不去哇

yanxin152133, 2024-08-28:
> > /etc/config/network 发出来
> 
> 没有更改过其它源码，只改默认IP，看起来是在不停重启网络，路由的网络接口灯一直一起亮又黑一会又亮一会，SSH上不去哇。当主路由用，用我没有编译无线驱动7915e，难道这个问题？

有没有可能你还需要`ip addr`看下你的网卡是哪个，再去配置默认ip和对应网卡？

yanxin152133, 2024-08-28:
编译电脑不仅可以被自动分配也可以设置静态ip

ramondelee, 2024-08-28:
好吧，检查了一下源码，上个月LEDE源码6088改了dtsi布局，应该是这个问题，话说改了适配哪个uboot？

ramondelee, 2024-09-03:
> > 好吧，检查了一下源码，上个月LEDE源码6088改了dtsi布局，应该是这个问题，话说改了适配哪个uboot？
> 
> 实测把上个月更新的内容还原就没问题，新的代码我试了几个uboot都一样

看了些说法，现在这个DTS布局是官方布局，已经抛弃第三方带webui的uboot了，忘记了官方是什么uboot了，我记得只能恢复刷机webui，是不是只是名字不一样，感觉没人说布局变了的操作方法

coolsnowwolf, 2026-09-05:
该问题对应的是 2024 年 TP-Link XDR 系列从传统 `fit` 分区切换到 `fitblk` 的迁移（提交 `1ae3c534669`）。该提交同时更新了 DTS 的 `root=/dev/fit0`、`fitblk` 升级逻辑和镜像包依赖；使用旧版第三方/厂商 U-Boot 或旧布局刷写新固件时，确实可能出现网络反复重启、DHCP 不可用等现象。

当前源码已按上游布局提供 `fitblk`，需要匹配支持该布局的 U-Boot 及对应升级流程；这不是 DHCP 或网卡驱动代码缺陷。issue 未提供当前提交、U-Boot 版本、完整 `.config` 或串口日志，无法安全制作兼容两种布局的通用补丁。请按 XDR6088 的官方 U-Boot/布局迁移说明操作；若使用匹配的新 U-Boot 仍失败，请重新提交完整日志。本 issue 关闭。

closed 2026-09-05 by coolsnowwolf

mirthlesslk, 2026-09-06:
目前使用的是天灵发布的uboot  https://www.right.com.cn/forum/thread-8328967-1-1.html，主要差别是target/linux/mediatek/dts/mt7986a-tplink-tl-xdr-common.dtsi  fip和ubi的范围变了，只需要更改这个范围仍可用旧版的uboot引导运行，目前使用的qwrt也是兼容的