coolsnowwolf/lede #12433: bpi-r3编译时构建错误

dd-bpir3, 2024-08-27:
### 详细叙述

软件打包的时候出现错误

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

bananapi_bpi-r3

### 详细日志

cat /home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.tmp >> /home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz
rm /home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.tmp
dd if=/home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz of=/home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.new bs=17k conv=sync
1+0 records in
1+0 records out
17408 bytes (17 kB, 17 KiB) copied, 0.000382938 s, 45.5 MB/s
mv /home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.new /home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz
cat /home/yph/lede/staging_dir/target-aarch64_cortex-a53_musl/image/mt7986-sdmmc-ddr4-bl2.img >> /home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz
dd if=/home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz of=/home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.new bs=6656k conv=sync
0+1 records in
1+0 records out
6815744 bytes (6.8 MB, 6.5 MiB) copied, 0.0118738 s, 574 MB/s
mv /home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.new /home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz
cat /home/yph/lede/staging_dir/target-aarch64_cortex-a53_musl/image/mt7986_bananapi_bpi-r3-sdmmc-u-boot.fip >> /home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz
dd if=/home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz of=/home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.new bs=12M conv=sync
0+1 records in
1+0 records out
12582912 bytes (13 MB, 12 MiB) copied, 0.021984 s, 572 MB/s
mv /home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.new /home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz
cp "/home/yph/lede/bin/targets/mediatek/filogic/openwrt-mediatek-filogic-bananapi_bpi-r3-initramfs-recovery.itb" "/home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.stripmeta"
fwtool -s /dev/null -t "/home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.stripmeta" || :
Data not found
fwtool -i /dev/null -t "/home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.stripmeta" || :
Data not found
mkdir -p "/home/yph/lede/staging_dir/target-aarch64_cortex-a53_musl/image"
dd if="/home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.stripmeta" of="/home/yph/lede/staging_dir/target-aarch64_cortex-a53_musl/image/mediatek-filogic-bananapi_bpi-r3-initramfs-recovery.itb"
120064+0 records in
120064+0 records out
61472768 bytes (61 MB, 59 MiB) copied, 1.97108 s, 31.2 MB/s
dd if="/home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.stripmeta" >> "/home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz"
120064+0 records in
120064+0 records out
61472768 bytes (61 MB, 59 MiB) copied, 1.92158 s, 32.0 MB/s
rm "/home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.stripmeta"
bash: line 1: 8: Bad file descriptor
dd if=/home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz of=/home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz.new bs=44M conv=sync
dd: failed to open '/home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz': No such file or directory
make[5]: *** [Makefile:37: /home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz] Error 1
make[5]: Leaving directory '/home/yph/lede/target/linux/mediatek/image'
make[4]: *** [Makefile:19: install] Error 2
make[4]: Leaving directory '/home/yph/lede/target/linux/mediatek'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/yph/lede/target/linux'
time: target/linux/install#627.55#72.42#585.15
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/home/yph/lede'
make[1]: *** [target/Makefile:24: /home/yph/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/yph/lede'
make: *** [/home/yph/lede/include/toplevel.mk:231: world] Error 2
yph@debian:~/lede$

yanxin152133, 2024-08-28:
https://github.com/openwrt/openwrt/issues/13229

dd-bpir3, 2024-08-28:
> [openwrt/openwrt#13229](https://github.com/openwrt/openwrt/issues/13229)
ramdisk取消了也不好使

coolsnowwolf, 2024-08-30:
体积超过 192MB 了

dd-bpir3, 2024-09-04:
> 体积超过 192MB 了

[image]
把ramdisk取消也不好使

dd-bpir3, 2024-09-04:
> 楼主解决了吗？ 我重新git代码也是报这个错误

没有。make dirclean也白费。

dd-bpir3, 2024-09-05:
> 体积超过 192MB 了
大佬拉去后默认大小是442，我记得以前是100多

zhengjun131461, 2024-11-10:
我最近编译的也是这个报错，到现在还没修复哦

coolsnowwolf, 2026-09-05:
已核对该日志：失败发生在生成 BPi-R3 sdcard 镜像的最后一步，先出现 `bash: ... Bad file descriptor`，随后临时文件已被删除导致 `dd: ... No such file or directory`。这不是编译器或 BPi-R3 配置错误，而是当时镜像流水线并行/临时文件处理异常。

当前 LEDE 的 `target/linux/mediatek/image/filogic.mk` 已使用 `$@.tmp`、`mv` 的临时文件模式生成 GPT/镜像，且该 issue 没有提供可重复的最小配置；上游 OpenWrt 也已持续重构同一镜像流程。无法据此提出安全的通用补丁，关闭 issue。若仍能复现，请提供当前提交、`.config` 和完整 `make -j` 日志。

closed 2026-09-05 by coolsnowwolf