coolsnowwolf/lede #12436: igc I226-V网卡驱动问题

lovexiaofan123, 2024-08-30:
### 详细叙述

我直接刷debian和win11就可以正确识别到I226-v网卡，也没有出现溢出和dropped增大。
用pve刷openwrt lede 网卡直通也没有出现溢出和dropped增大
但直接用硬盘直接刷入op lede会则会出现用lspci的日志
02:00.0 Ethernet controller: Intel Corporation Device 125c (rev 04)
03:00.0 Ethernet controller: Intel Corporation Device 125c (rev 04)
04:00.0 Ethernet controller: Intel Corporation Device 125c (rev 04)
05:00.0 Ethernet controller: Intel Corporation Device 125c (rev 04)
没有正确的识别到i226-v网卡信息，并且用ifconfig，查看所有连接的物理网口都会出现rx dropped值持续增大
已经测试多个内核版本，5.4 6.6 5.1 皆是如此
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

N5105 i226-v

### 详细日志

ifconfig
02:00.0 Ethernet controller: Intel Corporation Device 125c (rev 04)
03:00.0 Ethernet controller: Intel Corporation Device 125c (rev 04)
04:00.0 Ethernet controller: Intel Corporation Device 125c (rev 04)
05:00.0 Ethernet controller: Intel Corporation Device 125c (rev 04)

ZqinKing, 2024-09-04:
能驱动，显示125c是pci.ids信息没有更新吧。这个不影响使用。

lovexiaofan123, 2024-09-04:
> 能驱动，显示125c是pci.ids信息没有更新吧。这个不影响使用。

我看了源码的，看了i226的id识别是有的，但是不能识别到，识别问题还好，关键dropped值持续增大，lede任意op内核版本都存在，不是基于一个指定内核出问题
以及https://github.com/jim3ma/synology-igc/这个跟lede的igc源码是基本一致

coolsnowwolf, 2026-09-05:
已核对当前 LEDE：I226-V 的 PCI ID `0x125c` 已在 x86 `igc` 驱动中注册（包括 5.4 backport），`lspci` 显示“Device 125c”只是目标系统 `pci.ids` 数据库未提供设备名称，不代表驱动未识别。针对 I225/I226 掉线/丢包，5.10/5.15/6.1/6.6/6.12 等 x86 内核目录均已带有禁用 EEE 的补丁 `996-intel-igc-i225-i226-disable-eee.patch`。

该 issue 未提供 dmesg、ethtool、接口统计或可复现配置；讨论中也确认网卡可驱动，当前代码已有对应 EEE workaround。因此无需新增补丁，关闭 issue。若仍有持续 RX dropped，请附 `dmesg`、`ethtool -S`、链路协商和完整网络配置以便进一步定位。

closed 2026-09-05 by coolsnowwolf