coolsnowwolf/lede #12438: SSRP有点击有时候会报找到不LUCI的目录

ramondelee, 2024-08-30:
### 详细叙述

SSRP点击节点等标签页时，有时会出现某个标签（比如“访问控制”）反应较慢，在这个较慢的标签显示后，马上点击其它标签（比如“客户端”），就会出现这个报错，可以多试几次必现。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

TP XDR 6088

### 详细日志

Unable to launch the requested CGI program:
  /www/cgi-bin/luci: No such file or directory

ramondelee, 2024-08-30:
好吧，全网搜了一下自己解决了，暂时认为是SSRP的问题，解决方法如下：
在luci-ssr-plus文件中加入
set uhttpd.main.max_requests=50
commit uhttpd
之后，问题解决。

coolsnowwolf, 2026-09-05:
已核对当前 uhttpd 配置：默认 `max_requests` 为 3；该错误发生在并发/快速切换 LuCI 页面时，报告者已验证在 luci-app-ssr-plus 配置中设置 `uhttpd.main.max_requests=50` 并提交 UCI 后恢复正常。错误信息是 uhttpd 无法启动新的 `/www/cgi-bin/luci` 请求，并非 SSRP Lua 代码缺失。该参数应按设备性能和并发需求调整，不能由 SSRP 固定覆盖默认值，因此无需修改包代码。

根据已有复现后的解决确认，关闭 issue；若仍可复现，请附 uhttpd 日志和并发请求场景。

closed 2026-09-05 by coolsnowwolf