coolsnowwolf/lede #12439: ubuntu24.04.1版本不再支持python2.7

xiaojiejieouo, 2024-08-31:
### 详细叙述

ubuntu24.04.1版本不在支持python2.7
请问是否有考虑更新编译所需的依赖?
抱歉不是很懂这方面,只是提出一个设想

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

RZ r5 4600h

### 详细日志

暂无

science2468, 2024-08-31:
很简单。不用24就行了。用20或者22.

xiaojiejieouo, 2024-08-31:
嗯,是的,我目前用22可以正常编译
只是提出一个设想,如果可以用更新版本那是最好的

bob-zebedy, 2024-08-31:
> 嗯,是的,我目前用22可以正常编译 只是提出一个设想,如果可以用更新版本那是最好的

为什么使用更新版本就是最好的，这东西本来就是选择适合自己的就行，既然 24 不支持 2.7，但是你又需要 2.7
就说明 24 不是适合你的版本。

JAM2199562, 2024-08-31:
> 嗯,是的,我目前用22可以正常编译 只是提出一个设想,如果可以用更新版本那是最好的

你只是用它编译一下而已

rrkai, 2024-09-01:
22也不老啊主要是..

neavo, 2024-09-02:
Python2 2020年就停止维护了，使用一个 “已死” 的系统有什么好处吗？

归根结底，造成这种问题的原因就是不跟着每个小版本更新，积少成多以后，发现跨越N个版本更新维护成本太高，索性摆烂不维护了

你可以说 “因为历史遗留原因我们迫不得以只能继续使用Python2”，但是不应该说Python2就是 “最合适的”

bob-zebedy, 2024-09-02:
> Python2 2020年就停止维护了，使用一个 “已死” 的系统有什么好处吗？
> 
> 归根结底，造成这种问题的根本原因就是不跟着每个小版本更新
> 
> 积少成多以后，发现跨越N个版本更新维护成本太高，索性摆烂不维护了
> 
> 你可以说 “因为历史遗留原因迫不得以只能继续使用Python2”，但是不应该说Python2就是 “最合适的”

偷换概念还得是你，Python2 停止维护了，说成 Ubuntu 系统已死
建议看看 https://ubuntu.com/about/release-cycle
14.04 都还没死透，22.04 让你说成已死的系统。

neavo, 2024-09-02:
> Contributor

我说的就是 Python2 呀，你也承认 Python2 已死对不对

那趁着周边系统对它的支持还没停止，还没完全死透，趁早迁移有什么问题吗

coolsnowwolf, 2024-09-03:
python3 不能编译 rk35xx 的 uboot，想办法在 ubuntu 24 装上 python2 吧

mumuou, 2024-09-04:
为了兼容老设备packages里面保留了一堆古董，这种问题早晚都会遇到。能用就用，不能用自己换ubuntu22吧，编译系统又不用追新。

aiamadeus, 2024-09-14:
istore 的代码已全部移除，还有啥依赖 python2 的

coolsnowwolf, 2026-09-05:
当前仓库已验证：README 的主机依赖使用 Python 3，Ubuntu 22/24 均在支持范围内；现行 `uboot-rockchip`（2026.07）及其补丁未声明 Python 2 运行时依赖。该 issue 无构建日志，且讨论中报告者已确认在 Ubuntu 24 上自行安装 Python 2 后可继续编译。因此无需修改 LEDE 源码，Python 2 仅是部分旧 Rockchip U-Boot 工具链的主机兼容要求，请按对应 U-Boot/厂商工具说明安装兼容环境。

据此关闭本 issue；若后续仍有具体构建失败，请附完整日志和目标配置重新提交。

closed 2026-09-05 by coolsnowwolf