coolsnowwolf/lede #12440: 更新内核到6.1.106后编译报错

Kuron88, 2024-09-01:
### 详细叙述

/home/kuron/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915/ecm_interface.c: In function 'ecm_interface_list_stats_update':
/home/kuron/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915/ecm_interface.c:6759:25: error: implicit declaration of function 'br_dev_update_stats'; did you mean 'ppp_update_stats'? [-Werror=implicit-function-declaration]
 6759 |                         br_dev_update_stats(dev, &stats);
      |                         ^~~~~~~~~~~~~~~~~~~
      |                         ppp_update_stats
cc1: all warnings being treated as errors
make[5]: *** [scripts/Makefile.build:250: /home/kuron/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915/ecm_interface.o] Error 1
make[4]: *** [Makefile:2009: /home/kuron/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915] Error 2
make[4]: Leaving directory '/home/kuron/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/linux-6.1.106'
make[3]: *** [Makefile:164: /home/kuron/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915/.built] Error 2
make[3]: Leaving directory '/home/kuron/lede/package/qca/qca-nss-ecm'
time: package/qca/qca-nss-ecm/compile#15.14#3.07#18.21
    ERROR: package/qca/qca-nss-ecm failed to build.
make[2]: *** [package/Makefile:116: package/qca/qca-nss-ecm/compile] Error 1
make[2]: Leaving directory '/home/kuron/lede'
make[1]: *** [package/Makefile:110: /home/kuron/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kuron/lede'
make: *** [/home/kuron/lede/include/toplevel.mk:231：world] 错误 2

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

AX1800 PRO亚瑟

### 详细日志

/home/kuron/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915/ecm_interface.c: In function 'ecm_interface_list_stats_update':
/home/kuron/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915/ecm_interface.c:6759:25: error: implicit declaration of function 'br_dev_update_stats'; did you mean 'ppp_update_stats'? [-Werror=implicit-function-declaration]
 6759 |                         br_dev_update_stats(dev, &stats);
      |                         ^~~~~~~~~~~~~~~~~~~
      |                         ppp_update_stats
cc1: all warnings being treated as errors
make[5]: *** [scripts/Makefile.build:250: /home/kuron/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915/ecm_interface.o] Error 1
make[4]: *** [Makefile:2009: /home/kuron/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915] Error 2
make[4]: Leaving directory '/home/kuron/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/linux-6.1.106'
make[3]: *** [Makefile:164: /home/kuron/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915/.built] Error 2
make[3]: Leaving directory '/home/kuron/lede/package/qca/qca-nss-ecm'
time: package/qca/qca-nss-ecm/compile#15.14#3.07#18.21
    ERROR: package/qca/qca-nss-ecm failed to build.
make[2]: *** [package/Makefile:116: package/qca/qca-nss-ecm/compile] Error 1
make[2]: Leaving directory '/home/kuron/lede'
make[1]: *** [package/Makefile:110: /home/kuron/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kuron/lede'
make: *** [/home/kuron/lede/include/toplevel.mk:231：world] 错误 2

Kuron88, 2024-09-01:
这个报错解决了，又有新的报错，反正都是关于这个qca-nss-ecm的，太难搞了，退回旧的100内核，等着大佬去更新

Kuron88, 2024-09-01:
难怪大佬们一直不更新6.1的内核的，6.6的都更新几次了，可能是因为6.1更新报错的多，要修改nss的部分就没更新了

closed 2024-09-12 by Kuron88

smmya, 2024-10-26:
> 这个报错解决了，又有新的报错，反正都是关于这个qca-nss-ecm的，太难搞了，退回旧的100内核，等着大佬去更新

如何回退旧的100内核

jw10126121, 2024-12-13:
最新版，也遇到同样的错了，不知道怎么解决。

rsd1987, 2024-12-17:
> > 这个报错解决了，又有新的报错，反正都是关于这个qca-nss-ecm的，太难搞了，退回旧的100内核，等着大佬去更新
> 
> 如何回退旧的100内核

git checkout f9f1a684ea8143a24aabb264be4100a7f3aeb0d8

jw10126121, 2024-12-17:
CONFIG_PACKAGE_luci-app-ipsec-vpnd=y
CONFIG_PACKAGE_luci-i18n-ipsec-vpnd-zh-cn=y
我这边，这个插件，设置成y，就会出现这个错误。