coolsnowwolf/lede #12443: patch的全局vht判断，影响了7603中继WiFi6的速度

OpenFi-One, 2024-09-02:
### 详细叙述

package/kernel/mt76/patches-6.x/001-allow-vht-on-2g.patch
这个patch 由于加了全局的vht

7603是不支持vht的， 加上这个patch后导致7603 中继 wifi6的路由时出现协商到最低的6.5Mbit的速度
去掉这个patch后， 7603中继上级路由速度就正常了

建议加个条件判断 mt7603 不用vht

我在 OpenFi 5Pro 便携路由器上测试通过了
MT7621A + MT7613B + MT7603   内存512MB，  Flash 64MB

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

OpenFi 5Pro

### 详细日志

我在 OpenFi 5Pro 便携路由器上测试通过了， 去掉这个patch
MT7621A + MT7613B + MT7603   内存512MB，  Flash 64MB

OpenFi-One, 2024-09-02:
package/kernel/mt76/patches-5.4/001-allow-vht-on-2g.patch
package/kernel/mt76/patches/001-allow-vht-on-2g.patch
package/kernel/mt76/patches-6.x/001-allow-vht-on-2g.patch
建议这三个patch都加上对 MT7603的判断

因为我之前用便携路由 OpenFi 5Pro 

内核5.4 测试中继WiFi 6 路由速率低， 后来切换到5.10内核还是一样

对比后才发现是因为vht做了全局导致判断问题。

。

aiamadeus, 2024-09-03:
[image]
这个不是默认不开的吗

OpenFi-One, 2024-09-03:
是默认不开的， patch做了全局了，所以导致7603有问题。 如果7615开是可以的。

aiamadeus, 2024-09-03:
不是，你 vendor_vht (256qqam 不是 0？

OpenFi-One, 2024-09-03:
手动去配置没有试过是否可行， 
页面ui没看到选项，
可能是有判断7603不支持就没这个选项

我看补丁是全局判断的，估计有参数也不行。

aiamadeus, 2024-09-07:
你不开 256qam 这个补丁就不会生效...

OpenFi-One, 2024-09-08:
7603默认就是不开vht的。

coolsnowwolf, 2026-09-05:
已确认报告中的副作用并完成最小修复。MT7603 是仅 2.4GHz 的 11n 芯片，不支持 VHT；原 `001-allow-vht-on-2g.patch` 把传入的 `vht=true` 全局套用到 2.4GHz，导致 MT7603 被错误声明为 VHT，进而影响中继速率协商。

修复提交：`a76546030d7055af14abd8049a7b00e1bf5b73ac`（已推送 `coolsnowwolf/lede:master`）。四套内核补丁（5.4、5.10、6.x、6.18）统一改为仅当 `phy->cap.has_5ghz` 时才对 2.4GHz 开启 VHT；MT7603 保持关闭，MT7613/MT7615 等双频芯片继续支持 2.4GHz 256-QAM。

当前 x86_64/6.12.108 mt76 包已用 `-j30` 编译通过，补丁全部成功应用；`git diff --check` 也通过。请在 OpenFi 5Pro 刷入包含该提交的固件后复测中继速率。

closed 2026-09-05 by coolsnowwolf