coolsnowwolf/lede #12444: rockchip firmware 丢失

BenjaminX, 2024-09-03:
### 详细叙述

https://github.com/atf-builds
貌似都删了，怎么编译 R4s 等的固件呢？

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

R4S

### 详细日志

mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.atf-v2.8.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "atf-v2.8.tar.gz" "61df69619fd611da9e43abf66be28d6d59722feef559587fad0ca4cd9e499758" "" "https://github.com/atf-builds/atf/releases/download/v2.8/atf-v2.8.tar.gz?"    '
+ curl -f --connect-timeout 20 --retry 5 --location https://github.com/atf-builds/atf/releases/download/v2.8/atf-v2.8.tar.gz?/atf-v2.8.tar.gz

404 了

yanxin152133, 2024-09-03:
https://github.com/asytumn/atf/releases
是这个？

BenjaminX, 2024-09-03:
> https://github.com/asytumn/atf/releases 是这个？

换地址了？

yanxin152133, 2024-09-03:
> > https://github.com/asytumn/atf/releases 是这个？
> 
> 换地址了？

删库跑路？

yanxin152133, 2024-09-03:
package/boot/arm-trusted-firmware-rockchip/Makefile
[image]
PKG_SOURCE_URL:=https://github.com/asytumn/atf/releases/download/v$(PKG_VERSION)/atf-v$(PKG_VERSION).tar.gz?
改下试下？

BenjaminX, 2024-09-03:
> package/boot/arm-trusted-firmware-rockchip/Makefile [...]

我看看怎么改，不在本地编译的。在 action 上run的编译

BenjaminX, 2024-09-03:
lede/package/boot/arm-trusted-firmware-rockchip/Makefile

PKG_SOURCE:=atf-v$(PKG_VERSION).tar.gz
PKG_SOURCE_URL:=https://github.com/atf-builds/atf/releases/download/v$(PKG_VERSION)/atf-v$(PKG_VERSION).tar.gz?
PKG_HASH:=61df69619fd611da9e43abf66be28d6d59722feef559587fad0ca4cd9e499758

要召唤一下 lede 确认一下情况了，看起来是这里。

yanxin152133, 2024-09-03:
> lede/package/boot/arm-trusted-firmware-rockchip/Makefile
>
> PKG_SOURCE:=atf-v$(PKG_VERSION).tar.gz [...]

Pull requests过了，等等看：）

yanxin152133, 2024-09-03:
好像修复了，试下？

BenjaminX, 2024-09-04:
[image]
touch /workdir/openwrt/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.uboot-envtools_installed
echo "uboot-envtools" >> /workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/uboot-envtools.default.install
make[3]: Leaving directory '/workdir/openwrt/package/boot/uboot-envtools'
time: package/boot/uboot-envtools/compile#0.24#0.19#0.33
make[3]: Entering directory '/workdir/openwrt/package/boot/uboot-rk35xx'
rm -f /workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rockchip-rk3528-evb/uboot-rockchip-2023-07-25-32640b0a/.built
touch /workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rockchip-rk3528-evb/uboot-rockchip-2023-07-25-32640b0a/.built_check
make  -C /workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rockchip-rk3528-evb/uboot-rockchip-2023-07-25-32640b0a CROSS_COMPILE=aarch64-openwrt-linux-musl- DTC="/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.100/scripts/dtc/dtc" HOSTCC="gcc" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-aarch64_generic_musl/host/include -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-aarch64_generic_musl/host/include -std=gnu11" HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib -L/workdir/openwrt/staging_dir/hostpkg/lib -L/workdir/openwrt/staging_dir/target-aarch64_generic_musl/host/lib" LOCALVERSION="-OpenWrt-r6835-e90fe2d90" STAGING_PREFIX="/workdir/openwrt/staging_dir/host" PKG_CONFIG_PATH="/workdir/openwrt/staging_dir/host/lib/pkgconfig" PKG_CONFIG_LIBDIR="/workdir/openwrt/staging_dir/host/lib/pkgconfig" PKG_CONFIG_EXTRAARGS="--static" V='' IDB_SOC=rk3528 TPL_BIN=/workdir/openwrt/staging_dir/target-aarch64_generic_musl/image/rk3528_ddr_1056MHz_v1.06.bin u-boot.itb idbloader.img
make[4]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rockchip-rk3528-evb/uboot-rockchip-2023-07-25-32640b0a'
  CHK     include/config/uboot.release
  CHK     include/generated/version_autogenerated.h
  CHK     include/generated/timestamp_autogenerated.h
  CHK     include/generated/generic-asm-offsets.h
  CHK     include/generated/asm-offsets.h
start=$(aarch64-openwrt-linux-musl-nm u-boot | grep __rel_dyn_start | cut -f 1 -d ' '); end=$(aarch64-openwrt-linux-musl-nm u-boot | grep __rel_dyn_end | cut -f 1 -d ' '); tools/relocate-rela u-boot-nodtb.bin 0x00200000 $start $end
  COPY    u-boot.dtb
./"arch/arm/mach-rockchip/make_fit_atf.sh" \
arch/arm/dts/rk3528-evb.dtb > u-boot.its
/usr/bin/env: 'python2': No such file or directory
ls: cannot access 'bl31_0x*.bin': No such file or directory
  MKIMAGE u-boot.itb
Error: u-boot.its:46.29-30 syntax error
FATAL ERROR: Unable to parse input tree
./tools/mkimage: Can't read u-boot.itb.tmp: Invalid argument
make[4]: *** [Makefile:1056: u-boot.itb] Error 255
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rockchip-rk3528-evb/uboot-rockchip-2023-07-25-32640b0a'
make[3]: *** [Makefile:77: /workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rockchip-rk3528-evb/uboot-rockchip-2023-07-25-32640b0a/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/boot/uboot-rk35xx'
time: package/boot/uboot-rk35xx/rk3528-evb/compile#1.17#0.57#1.68
    ERROR: package/boot/uboot-rk35xx failed to build (build variant: rk3528-evb).
make[2]: *** [package/Makefile:116: package/boot/uboot-rk35xx/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

coolsnowwolf, 2024-09-04:
装上 Python2

closed 2024-09-04 by BenjaminX