coolsnowwolf/lede #12446: 请求将添加Fine3399

1967937965, 2024-09-03:
### 详细叙述

Fine3399是rumu3f设计的一个多用途嵌入式主板，采用bozztek的SW799系列核心板，处理器为RK3399。

Fine3399资料网站： https://gitee.com/opengisbook/Fine3399-Official

长风大佬的dts提交（其中有SPI Flash和屏幕的定义，但被注释了）： https://github.com/cm9vdA/build-linux/commit/ab3e4f56c5738e274b6200172816062648a44698

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

Fine3399

### 详细日志

无

coolsnowwolf, 2024-09-08:
已支持

1967937965, 2024-09-09:
fine3399编译报错，错误信息指出在执行dd命令时，无法找到rk3399-idbloader.bin文件，导致编译失败

1967937965, 2024-09-09:
/home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.100/arch/arm64/boot/dts/rockchip/rk3568-seewo-sv21.dts:515.20-519.4: Warning (avoid_default_addr_size): /usb@fd000000/device@2: Relying on default #size-cells value
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -DDTS_NO_LEGACY -I/home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.100/arch/arm64/boot/dts -I/home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.100/arch/arm64/boot/dts/include -I/home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.100/include/ -I/home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.100/scripts/dtc/include-prefixes -undef -D__DTS__  -o /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-orangepi-r1-plus-lts.dtb.tmp /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.100/arch/arm64/boot/dts/rockchip/rk3328-orangepi-r1-plus-lts.dts
/home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.100/scripts/dtc/dtc -O dtb -i/home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.100/arch/arm64/boot/dts/rockchip/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address    -o /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-guangmiao-g4c.dtb /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-guangmiao-g4c.dtb.tmp
rm -f /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-fine-3399.dtb.tmp
rm -f /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-h3399pc.dtb.tmp
rm -f /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-sv901-eaio.dtb.tmp
rm -f /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3568-seewo-sv21.dtb.tmp
/home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.100/scripts/dtc/dtc -O dtb -i/home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.100/arch/arm64/boot/dts/rockchip/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address    -o /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-orangepi-r1-plus-lts.dtb /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-orangepi-r1-plus-lts.dtb.tmp
/home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.100/scripts/dtc/dtc -O dtb -i/home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.100/arch/arm64/boot/dts/rockchip/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address    -o /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-orangepi-r1-plus.dtb /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-orangepi-r1-plus.dtb.tmp
rm -f /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-guangmiao-g4c.dtb.tmp
rm -f /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-orangepi-r1-plus-lts.dtb.tmp
rm -f /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-orangepi-r1-plus.dtb.tmp
mkdir -p /home/newuser/lede/bin/targets/rockchip/armv8 /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/tmp
rm -rf /home/newuser/lede/build_dir/target-aarch64_generic_musl/json_info_files
/home/newuser/lede/staging_dir/host/bin/mksquashfs4 /home/newuser/lede/build_dir/target-aarch64_generic_musl/root-rockchip /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/root.squashfs -nopad -noappend -root-owned -comp xz -Xpreset 9 -Xe -Xlc 0 -Xlp 2 -Xpb 2  -b 256k -p '/dev d 755 0 0' -p '/dev/console c 600 0 0 5 1' -no-xattrs
rm -f /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rumu3f_fine-3399-kernel.bin
cp /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/vmlinux /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rumu3f_fine-3399-kernel.bin
Parallel mksquashfs: Using 8 processors
Creating 4.0 filesystem on /home/newuser/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/root.squashfs, block size 262144.
Pseudo file "/dev" exists in source filesystem "/home/newuser/lede/build_dir/target-aarch64_generic_musl/root-rockchip/dev".
Ignoring, exclude it (-e/-ef) to override.
[... cut ...]
make[2]: Leaving directory '/home/newuser/lede'
make[1]: *** [target/Makefile:24: /home/newuser/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/newuser/lede'
make: *** [/home/newuser/lede/include/toplevel.mk:231: world] Error 2
newuser@RainYun-M0Eir52H:~/lede$ ls /home/newuser/lede/staging_dir/target-aarch64_generic_musl/image/
fine-3399-rk3399-uboot.img  loaderimage  rk322xh_bl31_v1.49.elf  rk3328-idbloader.bin  rk3328-trust.bin  rk3399_bl31.elf
newuser@RainYun-M0Eir52H:~/lede$

coolsnowwolf, 2024-09-12:
已修复

closed 2024-09-13 by coolsnowwolf

1967937965, 2024-09-16:
修复后，编译时找不到fine3399选项