coolsnowwolf/lede #12447: 远程桌面无法连接

weixieyou, 2024-09-04:
### 详细叙述

打开端口转发和动态ddns，有公网ip，无法访问内网的电脑，提示拒绝连接。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

r68s

### 详细日志

Wed Sep  4 14:01:57 2024 kern.warn kernel: [ 1568.474187] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:01:57 2024 kern.warn kernel: [ 1568.475075] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:01:57 2024 kern.warn kernel: [ 1568.475176] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:01:57 2024 kern.warn kernel: [ 1568.476721] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:01:57 2024 kern.warn kernel: [ 1568.479913] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:01:57 2024 kern.warn kernel: [ 1568.480395] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:01:57 2024 kern.warn kernel: [ 1568.481387] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:01:57 2024 kern.warn kernel: [ 1568.483606] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:01:57 2024 kern.warn kernel: [ 1568.493639] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:01:57 2024 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Wed Sep  4 14:01:58 2024 user.notice odhcpd: Reloading odhcpd due to ifup of wan_6 (pppoe-wan)
Wed Sep  4 14:02:00 2024 user.err ddns-scripts[12182]: 6: BusyBox nslookup error: '1'
Wed Sep  4 14:02:00 2024 user.err ddns-scripts[12184]: jjweixieeuorg: BusyBox nslookup error: '1'
Wed Sep  4 14:02:00 2024 user.err ddns-scripts[12183]: v6: BusyBox nslookup error: '1'
Wed Sep  4 14:02:01 2024 user.err ddns-scripts[12182]: 6: BusyBox nslookup error: '1'
Wed Sep  4 14:02:02 2024 user.err ddns-scripts[12183]: v6: BusyBox nslookup error: '1'
Wed Sep  4 14:02:02 2024 user.err ddns-scripts[12184]: jjweixieeuorg: BusyBox nslookup error: '1'
Wed Sep  4 14:02:02 2024 kern.warn kernel: [ 1573.477612] net_ratelimit: 521 callbacks suppressed
Wed Sep  4 14:02:02 2024 kern.warn kernel: [ 1573.477621] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:02:02 2024 kern.warn kernel: [ 1573.488106] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:02:02 2024 kern.warn kernel: [ 1573.490734] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:02:02 2024 kern.warn kernel: [ 1573.495030] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:02:02 2024 kern.warn kernel: [ 1573.497400] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:02:02 2024 kern.warn kernel: [ 1573.510909] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:02:02 2024 kern.warn kernel: [ 1573.512900] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:02:02 2024 kern.warn kernel: [ 1573.515389] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:02:02 2024 kern.warn kernel: [ 1573.521987] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:02:02 2024 kern.warn kernel: [ 1573.524437] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:02:02 2024 user.warn ddns-scripts[12182]: 6: Get registered/public IP for 'jj.weixiejigwei123.eu.org' failed - retry 1/0 in 60 seconds
Wed Sep  4 14:02:02 2024 user.warn ddns-scripts[12184]: jjweixieeuorg: Get registered/public IP for 'jj.weixie.eu.org' failed - retry 1/0 in 60 seconds
Wed Sep  4 14:02:02 2024 user.warn ddns-scripts[12183]: v6: Get registered/public IP for 'jjv6.weixiejigwei123.eu.org' failed - retry 1/0 in 60 seconds
Wed Sep  4 14:02:07 2024 kern.warn kernel: [ 1578.483433] net_ratelimit: 2029 callbacks suppressed
Wed Sep  4 14:02:07 2024 kern.warn kernel: [ 1578.483442] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:02:07 2024 kern.warn kernel: [ 1578.484433] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:02:07 2024 kern.warn kernel: [ 1578.486549] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:02:07 2024 kern.warn kernel: [ 1578.487435] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:02:07 2024 kern.warn kernel: [ 1578.489907] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:02:07 2024 kern.warn kernel: [ 1578.490146] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:02:07 2024 kern.warn kernel: [ 1578.496651] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:02:07 2024 kern.warn kernel: [ 1578.497001] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:02:07 2024 kern.warn kernel: [ 1578.499965] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:02:07 2024 kern.warn kernel: [ 1578.500167] br-lan: received packet on eth1 with own address as source address (addr:06:09:bb:b7:cf:35, vlan:0)
Wed Sep  4 14:03:06 2024 user.err ddns-scripts[12183]: v6: No or private or invalid IP 'fe80:0000:0000:0000:853b:62bc:d366:06a6' given! Please check your configuration
Wed Sep  4 14:03:06 2024 user.err ddns-scripts[12183]: v6: No update send to DDNS Provider
Wed Sep  4 14:06:03 2024 daemon.notice ttyd[17308]: [2024/09/04 14:06:03:5692] N: rops_handle_POLLIN_netlink: DELADDR
Wed Sep  4 14:06:03 2024 daemon.notice ttyd[17308]: [2024/09/04 14:06:03:5695] N: rops_handle_POLLIN_netlink: DELADDR
Wed Sep  4 14:06:03 2024 daemon.notice ttyd[17308]: [2024/09/04 14:06:03:5723] N: rops_handle_POLLIN_netlink: DELADDR
Wed Sep  4 14:06:06 2024 cron.err crond[1240]: crond (busybox 1.35.0) started, log level 5
Wed Sep  4 14:06:40 2024 cron.err crond[2728]: crond (busybox 1.35.0) started, log level 5
Wed Sep  4 14:06:43 2024 daemon.err modprobe: tun is already loaded
Wed Sep  4 14:07:01 2024 cron.err crond[5559]: crond (busybox 1.35.0) started, log level 5
Wed Sep  4 14:07:03 2024 daemon.err modprobe: tun is already loaded
Wed Sep  4 14:07:05 2024 daemon.notice ttyd[17308]: [2024/09/04 14:07:05:2058] N: rops_handle_POLLIN_netlink: DELADDR
Wed Sep  4 14:07:05 2024 daemon.notice ttyd[17308]: [2024/09/04 14:07:05:2061] N: rops_handle_POLLIN_netlink: DELADDR
Wed Sep  4 14:07:05 2024 daemon.notice ttyd[17308]: [2024/09/04 14:07:05:2112] N: rops_handle_POLLIN_netlink: DELADDR
Wed Sep  4 14:13:13 2024 user.warn ddns-scripts[12183]: v6: Updating IP at DDNS provider failed - starting retry 1/0
Wed Sep  4 14:13:15 2024 user.err ddns-scripts[12183]: v6: No or private or invalid IP 'fe80:0000:0000:0000:853b:62bc:d366:06a6' given! Please check your configuration
Wed Sep  4 14:13:15 2024 user.err ddns-scripts[12183]: v6: No update send to DDNS Provider
Wed Sep  4 14:23:17 2024 user.warn ddns-scripts[12183]: v6: Updating IP at DDNS provider failed - starting retry 2/0
Wed Sep  4 14:23:19 2024 user.err ddns-scripts[12183]: v6: No or private or invalid IP 'fe80:0000:0000:0000:853b:62bc:d366:06a6' given! Please check your configuration
Wed Sep  4 14:23:19 2024 user.err ddns-scripts[12183]: v6: No update send to DDNS Provider
Wed Sep  4 14:33:21 2024 user.warn ddns-scripts[12183]: v6: Updating IP at DDNS provider failed - starting retry 3/0
Wed Sep  4 14:33:23 2024 user.err ddns-scripts[12183]: v6: No or private or invalid IP 'fe80:0000:0000:0000:853b:62bc:d366:06a6' given! Please check your configuration
Wed Sep  4 14:33:23 2024 user.err ddns-scripts[12183]: v6: No update send to DDNS Provider
Wed Sep  4 14:36:51 2024 daemon.notice miniupnpd[12115]: SoapMethod: Unknown: GetFirewallStatus
Wed Sep  4 14:37:00 2024 daemon.notice miniupnpd[12115]: SoapMethod: Unknown: GetFirewallStatus

science2468, 2024-09-04:
检查下防火墙

yanxin152133, 2024-09-05:
这搞个rusdesk不比这个香嘛

coolsnowwolf, 2026-09-05:
日志没有显示远程桌面服务或端口转发本身的错误，反而明确显示：

- `br-lan: received packet on eth1 with own address as source address`，说明 LAN bridge/交换机/物理接线存在二层环路或接口重复加入；
- DDNS 的 IPv4/IPv6 多次 `BusyBox nslookup error: 1`，并把 `fe80::` 链路本地地址当作公网 IPv6，DDNS 更新必然失败。

请先断开可能的 LAN 回环、确认 eth1 只加入一个 bridge，关闭 TurboACC 流量分载后重载网络；再确认 DDNS 使用 wan/wan6 的全球地址（不能是 `fe80::`），最后从外网用 IP 直接测试端口。当前没有提供端口转发规则、目标主机监听状态或公网测试结果，因此无法归因于 LEDE 代码。按网络/第三方配置问题关闭；若仍存在，请附 `uci show firewall`、`uci show network`、目标主机监听端口和外网测试结果重新报告。

closed 2026-09-05 by coolsnowwolf