coolsnowwolf/lede #12448: photonicat

Namia-A, 2024-09-04:
### 详细叙述

6.1.6.6报错不知道能不能解决，在天灵那边能5.1编译

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

photonicat

### 详细日志

[image]

yanxin152133, 2024-09-05:
[image]

Namia-A, 2024-09-05:
> make V=s -j1
> 
> [image]
> 
> 

我用的gthub编译的

yanxin152133, 2024-09-05:
> 我用的gthub编译的

那就先学会Google 或者百度 再来发问题

yanxin152133, 2024-09-05:
翻了下你的GitHub，没找到有相关的编译记录
[image]

Namia-A, 2024-09-05:
> 翻了下你的GitHub，没找到有相关的编译记录
> 
> [image]
> 
> 

抱歉，我原始记录已经删了这个只是最开始的截图，我只是想知道这个错误的原理是什么，我才知道如何去改。

Namia-A, 2024-09-05:
> > 翻了下你的GitHub，没找到有相关的编译记录
> > 
> > [image]
> > 
> > 
> 
> 抱歉，我原始记录已经删了这个只是最开始的截图，我只是想知道这个错误的原理是什么，我才知道如何去改。

在求教个问题就是，大雕的群咧，被解散就找不到家了😂

closed 2024-09-08 by Namia-A