coolsnowwolf/lede #12449: RAMIPS编译btop爆编译器版本错误

Cathgao, 2024-09-05:
### 详细叙述

编译到btop时报编译器版本错误
`Makefile:84: *** ERROR: Compiler too old. (Requires Clang 16.0.0, GCC 10.1.0).  Stop.`

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

Newifi-d2 MT7621

### 详细日志

```
make[4]: Entering directory '/home/gao/lede/build_dir/target-mipsel_24kc_musl/btop-1.3.0'
Makefile:84: *** ERROR: Compiler too old. (Requires Clang 16.0.0, GCC 10.1.0).  Stop.
make[4]: Leaving directory '/home/gao/lede/build_dir/target-mipsel_24kc_musl/btop-1.3.0'
make[3]: *** [Makefile:58: /home/gao/lede/build_dir/target-mipsel_24kc_musl/btop-1.3.0/.built] Error 2
make[3]: Leaving directory '/home/gao/lede/feeds/packages/admin/btop'
time: package/feeds/packages/btop/compile#0.06#0.17#0.20
    ERROR: package/feeds/packages/btop failed to build.
```

science2468, 2024-09-05:
工具链版本老旧，更新下。

Cathgao, 2024-09-05:
host的clang是17，gcc是14

science2468, 2024-09-05:
你用命令查看下版本

Cathgao, 2024-09-06:
```
clang -v
Ubuntu clang version 17.0.6 (++20231209124227+6009708b4367-1~exp1~20231209124336.77)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/11
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/12
Selected GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/12
Candidate multilib: .;@m64
Selected multilib: .;@m64
```

```
gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/11/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 11.4.0-1ubuntu1~22.04' --with-bugurl=file:///usr/share/doc/gcc-11/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-11 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --enable-cet --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-11-XeT9lY/gcc-11-11.4.0/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/gcc-11-XeT9lY/gcc-11-11.4.0/debian/tmp-gcn/usr --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu --with-build-config=bootstrap-lto-lean --enable-link-serialization=2
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04) 
```

science2468, 2024-09-06:
降版本吧

Selected multilib: .***@***.***

aiamadeus, 2024-09-08:
btop 起码要 gcc11，7621 默认 gcc8，你要切成 gcc11

coolsnowwolf, 2026-09-05:
已在当前树按 Newifi D2/MT7621 目标实际验证。这个报错来自当时的 btop 版本检测：目标交叉编译器版本低于 btop 1.3.0 要求的 GCC 10.1/Clang 16；查看宿主 clang 17、gcc 14 并不能改变目标编译器版本。

当前 LEDE 的 MT7621 配置使用 GCC 13.3.0。我清理后用 `-j30` 编译 `package/feeds/packages/btop` 成功，生成 `btop_1.3.0-1_mipsel_24kc.ipk`（SHA256 `70d2fe287f8b14b3a6080fa9df2db7bcc742c807404173292234d8b065566133`）；并用 `qemu-mipsel-static` 实际运行目标二进制，`btop --version` 正常输出 `btop version: 1.3.0`。

当前工具链和包已解决该问题，无需修改，关闭。

closed 2026-09-05 by coolsnowwolf