coolsnowwolf/lede #12453: 请各位大神看一下 编译失败，卡这好久了

thylove-dsq, 2024-09-09:
### 详细叙述

请各位大神看一下 编译失败，卡这好久了  make[2]: *** [tools/Makefile:221: tools/tar/compile] Error 1

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

redmi ac2100

### 详细日志

wb@wb-virtual-machine:/file/router/lede$ sudo make V=s -j1
[sudo] wb 的密码： 
make[2]: Entering directory '/file/router/lede/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/file/router/lede/scripts/config'
make[1]: Entering directory '/file/router/lede'
....
....
....
checking whether mkdir handles trailing dot... yes
checking for mkdtemp... yes
checking for mkfifo... yes
checking whether mkfifo rejects trailing slashes... yes
checking whether mkfifoat rejects trailing slashes... yes
checking whether mknod can create fifo without root privileges... configure: error: in `/file/router/lede/build_dir/host/tar-1.34':
configure: error: you should not run configure as root (set FORCE_UNSAFE_CONFIGURE=1 in environment to bypass this check)
See `config.log' for more details
make[3]: *** [Makefile:36: /file/router/lede/build_dir/host/tar-1.34/.configured] Error 1
make[3]: Leaving directory '/file/router/lede/tools/tar'
time: tools/tar/compile#8.84#14.73#22.47
    ERROR: tools/tar failed to build.
make[2]: *** [tools/Makefile:221: tools/tar/compile] Error 1
make[2]: Leaving directory '/file/router/lede'
make[1]: *** [tools/Makefile:217: /file/router/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnyyynynyyyyyyynyynyynyynnynnyyyyyyyyyyyyyyyyynynnyyyyyyyy] Error 2
make[1]: Leaving directory '/file/router/lede'
make: *** [/file/router/lede/include/toplevel.mk:231：world] 错误 2

thylove-dsq, 2024-09-09:
用的不是root

bob-zebedy, 2024-09-11:
`错误：不应以 root 身份运行 configure（在环境中设置 FORCE_UNSAFE_CONFIGURE=1 可绕过此检查）`
翻译一下
错误：不应以 root 身份运行 configure（在环境中设置 FORCE_UNSAFE_CONFIGURE=1 可绕过此检查）

closed 2024-11-24 by aiamadeus