coolsnowwolf/lede #12455: 请大佬帮忙看下 编译失败

wanansany, 2024-09-11:
### 详细叙述

请帮忙看下什么原因

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86-64

### 详细日志

touch /opt/lede/staging_dir/target-x86_64_musl/stamp/.mosdns_installed
make[3]: Leaving directory '/opt/lede/feeds/packages/net/mosdns'
time: package/feeds/packages/mosdns/compile#59.18#12.84#26.84
make[3]: Entering directory '/opt/lede/feeds/helloworld/gn'
mkdir -p /opt/lede/dl
SHELL= flock /opt/lede/tmp/.gn-2024-08-13-54f5b539.tar.xz.flock -c '    /opt/lede/scripts/download.pl "/opt/lede/dl" "gn-2024-08-13-54f5b539.tar.xz" "1b5562417adfa29823301fea948197b23dfc887838fc94f96df5bbfd132dc592" "" || (   echo "Checking out files from the git repository..."; mkdir -p /opt/lede/tmp/dl && cd /opt/lede/tmp/dl && rm -rf gn-2024-08-13-54f5b539 && [ \! -d gn-2024-08-13-54f5b539 ] && git clone  https://gn.googlesource.com/gn.git gn-2024-08-13-54f5b539 && (cd gn-2024-08-13-54f5b539 && git checkout 54f5b539df8c4e460b18c62a11132d77b5601136 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd gn-2024-08-13-54f5b539 && git log -1 --format='\''@%ct'\''` && rm -rf gn-2024-08-13-54f5b539/.git &&         tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c gn-2024-08-13-54f5b539 |     xz -zc -7e > /opt/lede/tmp/dl/gn-2024-08-13-54f5b539.tar.xz && mv /opt/lede/tmp/dl/gn-2024-08-13-54f5b539.tar.xz /opt/lede/dl/ && rm -rf gn-2024-08-13-54f5b539;  )    '
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.cdn.openwrt.org/gn-2024-08-13-54f5b539.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   146    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.openwrt.org/gn-2024-08-13-54f5b539.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   146    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://mirror2.openwrt.org/sources/gn-2024-08-13-54f5b539.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   153    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
No more mirrors to try - giving up.
Checking out files from the git repository...
Cloning into 'gn-2024-08-13-54f5b539'...
remote: Total 33288 (delta 17527), reused 33288 (delta 17527)
Receiving objects: 100% (33288/33288), 34.74 MiB | 14.26 MiB/s, done.
Resolving deltas: 100% (17527/17527), done.
Note: switching to '54f5b539df8c4e460b18c62a11132d77b5601136'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 54f5b539 Markdown optimization (follow-up)
Packing checkout...
. /opt/lede/include/shell.sh; xzcat /opt/lede/dl/gn-2024-08-13-54f5b539.tar.xz | tar -C /opt/lede/build_dir/hostpkg/gn-2024-08-13-54f5b539/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /opt/lede/build_dir/hostpkg/gn-2024-08-13-54f5b539
touch /opt/lede/build_dir/hostpkg/gn-2024-08-13-54f5b539/.prepared76a891d820a9395ed6c9ed7777eca15c_6664517399ebbbc92a37c5bb081b5c53
python3 /opt/lede/build_dir/hostpkg/gn-2024-08-13-54f5b539/build/gen.py --no-last-commit-position
touch /opt/lede/build_dir/hostpkg/gn-2024-08-13-54f5b539/.configured
MAKEFLAGS="" /opt/lede/staging_dir/host/bin/ninja  -j1 -C /opt/lede/build_dir/hostpkg/gn-2024-08-13-54f5b539/out
ninja: Entering directory `/opt/lede/build_dir/hostpkg/gn-2024-08-13-54f5b539/out'
[1/295] CXX src/base/command_line.o
FAILED: src/base/command_line.o 
clang++ -MMD -MF src/base/command_line.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++20 -Wrange-loop-analysis -Wextra-semi-stmt -Wno-deprecated-copy -Wno-implicit-fallthrough -Wno-redundant-move -Wno-unused-variable -Wno-format -Wno-strict-aliasing -Wno-cast-function-type -c ../src/base/command_line.cc -o src/base/command_line.o
/bin/sh: 1: clang++: not found
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:58: /opt/lede/build_dir/hostpkg/gn-2024-08-13-54f5b539/.built] Error 1
make[3]: Leaving directory '/opt/lede/feeds/helloworld/gn'
time: package/feeds/helloworld/gn/host-compile#7.03#1.00#13.77
    ERROR: package/feeds/helloworld/gn [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/helloworld/gn/host/compile] Error 1
make[2]: Leaving directory '/opt/lede'
make[1]: *** [package/Makefile:110: /opt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/opt/lede'
make: *** [/opt/lede/include/toplevel.mk:231: world] Error 2

science2468, 2024-09-11:
404了，找找文件地址更换下。

Dawneng, 2024-09-11:
[gn-2024-08-13-54f5b539.tar.xz.zip](https://github.com/user-attachments/files/16966858/gn-2024-08-13-54f5b539.tar.xz.zip)
先解压一层 .zip  在复制到dl目录

qingtian110, 2024-09-12:
关键信息：`gn`，说明你编译固件时，选中了  naiveproxy  这个东西，但是你并不具备编译它的条件。

你有三个选择：
1、完善你的编译环境，至少满足 gn 编译的主机依赖
2、使用免编译源码  https://github.com/sbwml/openwrt_helloworld
3、不编译 naiveproxy

coolsnowwolf, 2026-09-05:
已确认日志中的源码镜像 404 不是最终失败：下载脚本随后成功从 GN Git 仓库检出并打包。首个真实错误是 `/bin/sh: clang++: not found`，属于宿主编译依赖缺失。

当前 README 的 Debian/Ubuntu 依赖列表已经明确包含 `clang`。我清理当前 `feeds/helloworld/gn` 后重新做 host 编译通过，生成的 `staging_dir/hostpkg/bin/gn` 可实际运行，版本输出为 `2518 (e8a8e0932a5e)`。

安装 README 列出的完整依赖（至少保证 `clang` 和 `clang++` 均在 PATH）即可；无需修改 GN 包，按环境问题关闭。

closed 2026-09-05 by coolsnowwolf