coolsnowwolf/lede #12456: 无线网口（联发科mt7925无法使用）

Kisoul, 2024-09-12:
### 详细叙述

编译过程选择了7925-commom、7925e、7925u 、wapd-openssl，无法在网络下无法看到“无线”选项，lspci可以看到此设备。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

mt7925

### 详细日志

root@OpenWrt:~# lspci
00:00.0 Host bridge: Intel Corporation 440FX - 82441FX PMC [Natoma] (rev 02)
00:01.0 ISA bridge: Intel Corporation 82371SB PIIX3 ISA [Natoma/Triton II]
00:01.1 IDE interface: Intel Corporation 82371SB PIIX3 IDE [Natoma/Triton II]
00:01.2 USB controller: Intel Corporation 82371SB PIIX3 USB [Natoma/Triton II] (rev 01)
00:01.3 Bridge: Intel Corporation 82371AB/EB/MB PIIX4 ACPI (rev 03)
00:02.0 VGA compatible controller: Device 1234:1111 (rev 02)
00:03.0 Unclassified device [00ff]: Red Hat, Inc. Virtio memory balloon
00:05.0 PCI bridge: Red Hat, Inc. QEMU PCI-PCI bridge
00:07.0 SATA controller: Intel Corporation 82801IR/IO/IH (ICH9R/DO/DH) 6 port SATA Controller [AHCI mode] (rev 02)
00:10.0 Network controller: MEDIATEK Corp. Device 7925
00:11.0 Ethernet controller: Intel Corporation Device 125c (rev 04)
00:12.0 Ethernet controller: Red Hat, Inc. Virtio network device
00:1b.0 Ethernet controller: Intel Corporation Device 125c (rev 04)
00:1c.0 Ethernet controller: Intel Corporation Device 125c (rev 04)
00:1e.0 PCI bridge: Red Hat, Inc. QEMU PCI-PCI bridge
00:1f.0 PCI bridge: Red Hat, Inc. QEMU PCI-PCI bridge
03:0d.0 Ethernet controller: Intel Corporation Device 125c (rev 04)
03:0e.0 Ethernet controller: Intel Corporation Device 125c (rev 04)
root@OpenWrt:~#

aiamadeus, 2024-09-14:
2fdd47aff023271f938dc1c4b0982c7ca705e8c9 要选 mt7925-firmware

Kisoul, 2024-09-14:
> [2fdd47a](https://github.com/coolsnowwolf/lede/commit/2fdd47aff023271f938dc1c4b0982c7ca705e8c9) 要选 mt7925-firmware

CONFIG_PACKAGE_kmod-mt7921-common=y
CONFIG_PACKAGE_kmod-mt7921-firmware=y
CONFIG_PACKAGE_kmod-mt7921e=y
CONFIG_PACKAGE_kmod-mt7921s=y
CONFIG_PACKAGE_kmod-mt7921u=y
CONFIG_PACKAGE_kmod-mt7922-firmware=y
CONFIG_PACKAGE_kmod-mt7925-common=y
CONFIG_PACKAGE_kmod-mt7925e=y
CONFIG_PACKAGE_kmod-mt7925u=y
CONFIG_PACKAGE_kmod-mt792x-common=y
CONFIG_PACKAGE_kmod-mt792x-usb=y

截止到9月8日config重中没有这个选项

aiamadeus, 2024-09-15:
你就不能拉最新的重新编啊。。

Kisoul, 2024-09-16:
> 你就不能拉最新的重新编啊。。

已重新拉取并解决

closed 2024-09-16 by Kisoul

biuyoudeaad, 2024-10-10:
大佬编译好了，为啥我的wifi 的热点的信号0 噪声0哇，有遇到吗？
是我编译时选择的不对吗
[image]

biuyoudeaad, 2024-10-10:
进入wifi设置是这样
[image]

IDeLoveYou, 2024-10-25:
6ghz 不能用，5ghz 可以，但是好像不支持双频并发

biuyoudeaad, 2024-10-25:
哦哦，谢谢大佬。我说为啥编译进去识别了，只要开6ghz的就不行。

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年10月25日 08:53 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [coolsnowwolf/lede] 无线网口（联发科mt7925无法使用） (Issue #12456) |

6ghz 不能用，5ghz 可以，但是好像不支持双频并发

aiamadeus, 2024-11-07:
6GHz 需要设置 wpa3 加密，你们试试？

IDeLoveYou, 2024-11-07:
> 6GHz 需要设置 wpa3 加密，你们试试？

哥们，能看看你是怎么配的吗，我就用的wpa3

apple-2025-us, 2025-06-14:
请问怎么编译7925驱动？我的和你一样，不能识别。

IDeLoveYou, 2025-06-14:
我使用勾上这些后就可以用了
CONFIG_DEFAULT_kmod-mt7921e=y
CONFIG_PACKAGE_kmod-mt7921-common=y
CONFIG_PACKAGE_kmod-mt7921-firmware=y
CONFIG_PACKAGE_kmod-mt7921e=y

IDeLoveYou, 2025-06-14:
进入编译的界面，按反斜杆，可以搜索，搜索关键词，找到对应插件，按空格，M 就是单独编译成 ipk 包，*就是编译进固件

apple-2025-us, 2025-06-14:
编译界面有连接吗？你能帮我编译个驱动吗？我是openwrt24.10.0的R5C路由谢谢

IDeLoveYou, 2025-06-14:
额，你没编译过固件吗？自己琢磨琢磨吧，这也是学习的过程，我没精力帮忙编译的，你可以找别人帮你编译，但我还是建议你自己学一下

apple-2025-us, 2025-06-14:
我主要不知道哪个是编译的界面。英文看不懂哈

IDeLoveYou, 2025-06-14:
编译的时候不是通过make menuconfig 选择需要的插件吗，就这个界面

apple-2025-us, 2025-06-14:
> IDeLoveYou  left a comment  (coolsnowwolf/lede#12456) (https://github.com/coolsnowwolf/lede/issues/12456#issuecomment-2972851314)
>
> 编译的时候不是通过make menuconfig 选择需要的插件吗，就这个界面

主要是找不到编译的网址