coolsnowwolf/lede #12459: 新版本的固件不能上网,包括6.6.50,6.1.100

guhill1, 2024-09-13:
### 详细叙述

新版本的固件不能上网,架构是x86,包括6.6.50,6.1.100,之前相同配置可以上网,
查看系统日志显示kernel ipset segment fault, xx xx xx xx 数字.
貌似和turbo acc shortcut FE 或者 kmod-fast-classifier 有关,请大佬看看怎么回事.
现在刷9月8日的版本可以正常上网.

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86-64

### 详细日志

系统日志显示
kernel ipset segment fault, xx xx xx xx 数字.

eearphon, 2024-09-13:
root@OpenWrt:~# cat /etc/os-release 
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r6843-82ffc84e2"
OPENWRT_BOARD="x86/64"
OPENWRT_ARCH="x86_64"
OPENWRT_TAINTS="no-all busybox"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r6843-82ffc84e2"

同样log：
ipset[25428]: segfault at 599e7ef8 ip 00007fc6b4f6267a sp 00007ffe599e6c78 error 4 in libc.so[7fc6b4f1d000+55000] likely on CPU 
3 (core 0, socket 6)

eearphon, 2024-09-14:
晕，一下子回滚重来这么多啊

guhill1, 2024-09-21:
> 晕，一下子回滚重来这么多啊

现在编译是没问题了

closed 2024-09-21 by guhill1