coolsnowwolf/lede #12460: macos m2下编译树莓派5固件失败

netahui, 2024-09-13:
### 详细叙述

macos m2下编译树莓派5固件失败

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

macos m2 编译树莓派5固件

### 详细日志

clang: warning: argument unused during compilation: '-no-pie' [-Wunused-command-line-argument]
ld: warning: ignoring duplicate libraries: '../libcpp/libcpp.a', '../libdecnumber/libdecnumber.a', 'libcommon.a'
Undefined symbols for architecture arm64:
  "_ZSTD_compress", referenced from:
      lto_end_compression(lto_compression_stream*) in libbackend.a[225](lto-compress.o)
  "_ZSTD_compressBound", referenced from:
      lto_end_compression(lto_compression_stream*) in libbackend.a[225](lto-compress.o)
  "_ZSTD_decompress", referenced from:
      lto_end_uncompression(lto_compression_stream*, lto_compression) in libbackend.a[225](lto-compress.o)
  "_ZSTD_getErrorName", referenced from:
      lto_end_compression(lto_compression_stream*) in libbackend.a[225](lto-compress.o)
      lto_end_uncompression(lto_compression_stream*, lto_compression) in libbackend.a[225](lto-compress.o)
  "_ZSTD_getFrameContentSize", referenced from:
      lto_end_uncompression(lto_compression_stream*, lto_compression) in libbackend.a[225](lto-compress.o)
  "_ZSTD_isError", referenced from:
      lto_end_compression(lto_compression_stream*) in libbackend.a[225](lto-compress.o)
      lto_end_uncompression(lto_compression_stream*, lto_compression) in libbackend.a[225](lto-compress.o)
  "_ZSTD_maxCLevel", referenced from:
      lto_end_compression(lto_compression_stream*) in libbackend.a[225](lto-compress.o)
      lto_end_compression(lto_compression_stream*) in libbackend.a[225](lto-compress.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [/Volumes/TEMP/lede/build_dir/toolchain-aarch64_cortex-a76_gcc-12.3.0_musl/gcc-12.3.0/gcc/c/Make-lang.in:87: cc1] Error 1
make[5]: Leaving directory '/Volumes/TEMP/lede/build_dir/toolchain-aarch64_cortex-a76_gcc-12.3.0_musl/gcc-12.3.0-initial/gcc'
make[4]: *** [Makefile:4597: all-gcc] Error 2
make[4]: Leaving directory '/Volumes/TEMP/lede/build_dir/toolchain-aarch64_cortex-a76_gcc-12.3.0_musl/gcc-12.3.0-initial'
make[3]: *** [Makefile:37: /Volumes/TEMP/lede/build_dir/toolchain-aarch64_cortex-a76_gcc-12.3.0_musl/gcc-12.3.0-initial/.built] Error 2
make[3]: Leaving directory '/Volumes/TEMP/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#0.64#0.37#0.94
    ERROR: toolchain/gcc/initial failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/gcc/initial/compile] Error 1
make[2]: Leaving directory '/Volumes/TEMP/lede'
make[1]: *** [toolchain/Makefile:93: /Volumes/TEMP/lede/staging_dir/toolchain-aarch64_cortex-a76_gcc-12.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/Volumes/TEMP/lede'
make: *** [/Volumes/TEMP/lede/include/toplevel.mk:231: world] Error 2

netahui, 2024-09-14:
macos（Macbook pro m2）还是不行，最后在pc装了一个ubuntu，编译成功

netahui, 2024-09-14:
烦表大神帮助解决一下，macos 14.6.1（Macbook pro m2）如何才能编译成功

tckxjzc, 2025-02-25:
macos用虚拟机装个ubuntu就可以了

coolsnowwolf, 2026-09-05:
已核对当前工具链实现。日志的直接原因是 macOS/arm64 链接 GCC 初始阶段的 `cc1` 时没有带入 zstd，因而所有 `ZSTD_*` 符号未解析，并非树莓派 5 目标代码错误。

当前树的 GCC 配置已显式使用构建系统自己的 zstd：

- `ac_cv_search_ZSTD_compress=$(STAGING_DIR_HOST)/lib/libzstd.a -pthread`，用绝对路径保证不会误用 Homebrew/系统版本；
- `--with-zstd=$(STAGING_DIR_HOST)`；
- `tools/Makefile` 保证其他 host 工具依赖本树的 zstd。

这覆盖了日志中的链接缺失路径。原报告者也已经确认改用 Ubuntu 后完整编译成功。由于报告基于 2024 年旧构建树，且没有更新后的 macOS 日志或可复现配置，当前无需追加补丁，按过期且已有修复路径关闭。macOS 原生重试时请严格按当前 README 安装 GNU 工具并使用大小写敏感文件系统；若当前 HEAD 仍失败，请清空旧 `staging_dir`、`build_dir/toolchain-*` 和 ccache 后提供首个失败命令的 `V=s` 完整链接行再开新 issue。

closed 2026-09-05 by coolsnowwolf