coolsnowwolf/lede #12461: toolchain/gcc切换到12后ssrp运行不正常

monsterzzzz, 2024-09-13:
### 详细叙述

在原来的11版本下ssrp是正常的，toolchain/gcc切换到12后整体编译能够通过，但ssrp运行不正常，如果在命令行运行，会出现Segmentation fault，设置为全局SOCKS5是正常的，说明本身节点正常
推测编译出来的二进制文件有问题

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

R68S

### 详细日志

root@OPR68S:~# /etc/init.d/shadowsocksr start
Segmentation fault
Segmentation fault
ssr-rules[18232]: Start failed!

CYaNu, 2024-09-13:
passwall2 一样段错误

monsterzzzz, 2024-09-13:
回退到切换toolchain/gcc前的commit就正常了，不知道具体是哪个环节的问题

monsterzzzz, 2024-09-13:
也有可能是toolchain/musl问题

CYaNu, 2024-09-13:
gcc我切回12依然有问题 不是gcc的锅

monsterzzzz, 2024-09-13:
我回到gcc11就没问题了
cceaa09ba2e21463246f82c3d13e00fc97af500e 这个commit

CYaNu, 2024-09-13:
reset中

monsterzzzz, 2024-09-13:
貌似找到原因了，是ipset出了问题，执行时直接报错
root@OPR68S:~# ipset
Segmentation fault

CYaNu, 2024-09-13:
[image]

CYaNu, 2024-09-13:
已经知道了

monsterzzzz, 2024-09-13:
换成官方源的ipset应该就可以了，多了一个patch

qingtian110, 2024-09-13:
我 GCC 15 Snapshot 了

[image]

CYaNu, 2024-09-14:
已确认是lede的ipset版本太老所致 起码在6.6.50下有问题 更新openwrt管源ipset加patch编译最新commit没问题

WongKwongYu, 2024-09-14:
不用重新编译，登录路由器，软件包里面ipset更新到新版就行了，ssrp和passwall都正常了

coolsnowwolf, 2026-09-05:
已确认并验证修复。问题不在 SSR Plus / Passwall，而是旧 ipset 在 musl 1.2.5 下使用了错误的 `basename()` 声明，导致运行时段错误。仓库已于 2024-09-14 合入 `ipset: include libgen.h for basename`，随后持续更新 ipset。

当前版本为 ipset 7.24。我清理后用 `-j30` 重新编译成功，并使用目标 x86_64 musl loader 实际运行产物：`ipset --version` 正常输出 `ipset v7.24, protocol version: 7`，`ipset help` 正常返回完整帮助，无 Segmentation fault。产物为 `ipset_7.24-1_x86_64.ipk`，SHA256 `b32d5d099d166c5b08570d6a9dc9b532b610989290a23de595162fc99666e81b`。

因此当前树已解决该问题，关闭。

closed 2026-09-05 by coolsnowwolf