coolsnowwolf/lede #12462: 编译失败

GreatMichaelLee, 2024-09-13:
### 详细叙述

编译报错，详见日志。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

touch /home/builder/lede_x86/build_dir/target-x86_64_musl/libbpf-1.4.5/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/libbpf-1.4.5=libbpf-1.4.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro    -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/usr/include -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/include/fortify -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/include   " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/libbpf-1.4.5=libbpf-1.4.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro    -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/usr/include -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/include/fortify -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/include   " LDFLAGS="-L/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/usr/lib -L/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/lib -znow -zrelro   " make  -C /home/builder/lede_x86/build_dir/target-x86_64_musl/libbpf-1.4.5/src AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/libbpf-1.4.5=libbpf-1.4.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  " LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" V=1 LIBSUBDIR=lib ;
make[4]: Entering directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/libbpf-1.4.5/src'
x86_64-openwrt-linux-musl-gcc -I. -I../include -I../include/uapi -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/libbpf-1.4.5=libbpf-1.4.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro    -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/usr/include -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/include/fortify -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/include    -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wno-unknown-warning-option -Wno-format-overflow  -I/home/builder/lede_x86/staging_dir/target-x86_64_musl/usr/include   -c btf.c -o staticobjs/btf.o
In file included from btf.c:18:
/home/builder/lede_x86/staging_dir/target-x86_64_musl/usr/include/gelf.h:86:9: error: unknown type name 'Elf64_Relr'
   86 | typedef Elf64_Relr GElf_Relr;
      |         ^~~~~~~~~~
cc1: note: unrecognized command-line option '-Wno-unknown-warning-option' may have been intended to silence earlier diagnostics
make[4]: *** [Makefile:134: staticobjs/btf.o] Error 1
make[4]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/libbpf-1.4.5/src'
make[3]: *** [Makefile:75: /home/builder/lede_x86/build_dir/target-x86_64_musl/libbpf-1.4.5/.built] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/package/libs/libbpf'
time: package/libs/libbpf/compile#0.05#0.12#0.16
    ERROR: package/libs/libbpf failed to build.
make[2]: *** [package/Makefile:116: package/libs/libbpf/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2
builder@Build-Server:/home/builder/lede_x86$

qingtian110, 2024-09-13:
musl1.2.5 + gcc13 + libbpf1.4.5

所以没有一个人觉得这个配方奇怪么，埋头就 build

love-dy, 2024-09-13:
> musl1.2.5 + gcc13 + libbpf1.4.5
> 
> 所以没有一个人觉得这个配方奇怪么，埋头就 build

哈，我突然想到一个词，缝合

ZGNCZMY, 2024-09-13:
> musl1.2.5 + gcc13 + libbpf1.4.5
> 
> 所以没有一个人觉得这个配方奇怪么，埋头就 build

哈哈，好久没更新系统了想更新下

GreatMichaelLee, 2024-09-14:
> musl1.2.5 + gcc13 + libbpf1.4.5
> 
> 所以没有一个人觉得这个配方奇怪么，埋头就 build

听不懂，不是每一个人都有相同的知识水平，请教不埋头build,应该如何？

GreatMichaelLee, 2024-09-14:
强制rebase基线后，现在出这个了，咋弄，这冲突是个啥啊

Collected errors:
 * check_data_file_clashes: Package kmod-r8125-rss wants to install file /home/builder/lede_x86/build_dir/target-x86_64_musl/root-x86/lib/modules/6.6.50/r8125.ko
        But that file is already provided by package  * kmod-r8125
 * opkg_install_cmd: Cannot install package kmod-r8125-rss.
 * check_data_file_clashes: Package kmod-r8126-rss wants to install file /home/builder/lede_x86/build_dir/target-x86_64_musl/root-x86/lib/modules/6.6.50/r8126.ko
        But that file is already provided by package  * kmod-r8126
 * opkg_install_cmd: Cannot install package kmod-r8126-rss.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:111: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2
builder@Build-Server:/home/builder/lede_x86$

aiamadeus, 2024-09-14:
去掉 r8125，改选带 rss 的

HollyMadCow, 2024-09-15:
> 强制rebase基线后，现在出这个了，咋弄，这冲突是个啥啊
>
> Collected errors:
>
> * check_data_file_clashes: Package kmod-r8125-rss wants to install file [...]

删除config文件，重新make menuconfig

coolsnowwolf, 2026-09-05:
已按当前树分别验证两个报错。

1. 最初的 `Elf64_Relr` 报错是更新源码后 staging 中的 ELF 头文件版本不一致：`gelf.h` 引用了新类型，但实际使用了没有该定义的旧 `elf.h`。当前树已是 libbpf 1.7.0，当前 staging 的 `libelf.h` / `gelf.h` 类型匹配；我清理 libbpf 后用 `-j30` 重新编译成功，产物 `libbpf1_1.7.0-1_x86_64.ipk`，SHA256 为 `d411fc25d1ccc3a69d3036a8b51dc785d635b1a5e129f61ae2aaf19a0f557ba6`。遇到这种跨版本残留应清理对应包或重新建立 staging，不要只强制 rebase 后沿用旧构建目录。

2. 后续 `r8125.ko` / `r8126.ko` 文件冲突是普通版与 RSS 版被同时选中。当前包已明确声明 `kmod-r8125-rss` 冲突 `kmod-r8125`、`kmod-r8126-rss` 冲突 `kmod-r8126`，生成的 IPK 控制信息也包含该 `Conflicts` 字段；每组只能选择一个。当前 x86 配置只选择 RSS 版。

当前代码和干净配置下均已解决，不需要追加补丁，关闭。

closed 2026-09-05 by coolsnowwolf