coolsnowwolf/lede #12464: tini编译报错，dockerd依赖项

binge8, 2024-09-13:
### 详细叙述

actions全新环境编译，config也试过重新获取，还是报错。好像是dockerd的依赖项

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x64，armv8

### 详细日志

```bash
2024-09-13T13:32:26.4318091Z make[3]: Entering directory '/mnt/openwrt/feeds/packages/utils/tini'
2024-09-13T13:32:26.5742904Z rm -f /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/tini-0.19.0/.built
2024-09-13T13:32:26.5772525Z touch /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/tini-0.19.0/.built_check
2024-09-13T13:32:26.5799184Z MAKEFLAGS="" /home/runner/work/op/op/openwrt/staging_dir/host/bin/ninja  -j1 -C /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/tini-0.19.0 
2024-09-13T13:32:26.5831297Z ninja: Entering directory `/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/tini-0.19.0'
2024-09-13T13:32:26.6040095Z [1/4] Building C object CMakeFiles/tini.dir/src/tini.c.o
2024-09-13T13:32:26.6040924Z FAILED: CMakeFiles/tini.dir/src/tini.c.o 
2024-09-13T13:32:26.6046978Z /home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc  -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/tini-0.19.0 -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/tini-0.19.0=tini-0.19.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -std=gnu99 -Werror -Wextra -Wall -pedantic-errors -O2 -fstack-protector --param=ssp-buffer-size=4 -Wformat -DNDEBUG -MD -MT CMakeFiles/tini.dir/src/tini.c.o -MF CMakeFiles/tini.dir/src/tini.c.o.d -o CMakeFiles/tini.dir/src/tini.c.o -c /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/tini-0.19.0/src/tini.c
2024-09-13T13:32:26.6053539Z /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/tini-0.19.0/src/tini.c: In function 'print_usage':
2024-09-13T13:32:26.6055308Z /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/tini-0.19.0/src/tini.c:227:36: error: implicit declaration of function 'basename' [-Wimplicit-function-declaration]
2024-09-13T13:32:26.6056436Z   227 |         fprintf(file, "%s (%s)\n", basename(name), TINI_VERSION_STRING);
2024-09-13T13:32:26.6056925Z       |                                    ^~~~~~~~
2024-09-13T13:32:26.6058369Z /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/tini-0.19.0/src/tini.c:227:25: error: format '%s' expects argument of type 'char *', but argument 3 has type 'int' [-Werror=format=]
2024-09-13T13:32:26.6059514Z   227 |         fprintf(file, "%s (%s)\n", basename(name), TINI_VERSION_STRING);
2024-09-13T13:32:26.6060010Z       |                        ~^          ~~~~~~~~~~~~~~
2024-09-13T13:32:26.6060382Z       |                         |          |
2024-09-13T13:32:26.6060916Z       |                         char *     int
2024-09-13T13:32:26.6061417Z       |                        %d
2024-09-13T13:32:26.6063225Z /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/tini-0.19.0/src/tini.c:232:32: error: format '%s' expects argument of type 'char *', but argument 3 has type 'int' [-Werror=format=]
2024-09-13T13:32:26.6065179Z   232 |         fprintf(file, "Usage: %s [OPTIONS] PROGRAM -- [ARGS] | --version\n\n", basename(name));
2024-09-13T13:32:26.6066030Z       |                               ~^                                               ~~~~~~~~~~~~~~
2024-09-13T13:32:26.6066646Z       |                                |                                               |
2024-09-13T13:32:26.6067275Z       |                                char *                                          int
2024-09-13T13:32:26.6067843Z       |                               %d
2024-09-13T13:32:26.6069551Z /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/tini-0.19.0/src/tini.c:234:90: error: format '%s' expects argument of type 'char *', but argument 3 has type 'int' [-Werror=format=]
2024-09-13T13:32:26.6071920Z   234 |         fprintf(file, "Execute a program under the supervision of a valid init process (%s)\n\n", basename(name));
2024-09-13T13:32:26.6073270Z       |                                                                                         ~^        ~~~~~~~~~~~~~~
2024-09-13T13:32:26.6073958Z       |                                                                                          |        |
2024-09-13T13:32:26.6074711Z       |                                                                                          char *   int
2024-09-13T13:32:26.6075209Z       |                                                                                         %d
2024-09-13T13:32:26.6075866Z cc1: all warnings being treated as errors
2024-09-13T13:32:26.6076511Z ninja: build stopped: subcommand failed.
2024-09-13T13:32:26.6077444Z make[3]: *** [Makefile:45: /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/tini-0.19.0/.built] Error 1
2024-09-13T13:32:26.6078284Z make[2]: *** [package/Makefile:116: package/feeds/packages/tini/compile] Error 1
2024-09-13T13:32:26.6079279Z make[1]: *** [package/Makefile:110: /home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2024-09-13T13:32:26.6080424Z make[3]: Leaving directory '/mnt/openwrt/feeds/packages/utils/tini'
```

ZGNCZMY, 2024-09-13:
也是同样的报错

love-dy, 2024-09-13:
大佬，这是又更新啥了，同样报错

Look-UI, 2024-09-14:
周四编译的时候就报错，请大佬更新支持

zxlhhyccc, 2024-09-14:
添加这个补丁应该可以解决你的问题。新建补丁名字：002-Support-POSIX-basename-from-musl-libc.patch，然后把这些代码复制粘贴进去即可。
补丁放到：feeds/packages/utils/tini/patches/ 目录下。

love-dy, 2024-09-14:
大佬们，有解决方案了吗，可以开始编译了吗

zxlhhyccc, 2024-09-14:
> 大佬们，问题解决了吗，可以开始编译了吗

加上这个补丁即可编译。

love-dy, 2024-09-14:
> > 大佬们，问题解决了吗，可以开始编译了吗
> 
> 加上这个补丁即可编译。都已经提了pr。

嗯，大佬，你的办法看上去不错，可我是action编译的，很多Linux指令还不太会，还是等源码修复再编译吧

ZGNCZMY, 2024-09-14:
> > > 大佬们，问题解决了吗，可以开始编译了吗
> > 
> > 
> > 加上这个补丁即可编译。都已经提了pr。
> 
> 嗯，大佬，你的办法看上去不错，可我是action编译的，很多Linux指令还不太会，还是等源码修复再编译吧

基础的要自己学，饭来了还要人端不可取......

love-dy, 2024-09-14:
> > > > 大佬们，问题解决了吗，可以开始编译了吗
> > > 
> > > 
> > > 加上这个补丁即可编译。都已经提了pr。
> > 
> > 
> > 嗯，大佬，你的办法看上去不错，可我是action编译的，很多Linux指令还不太会，还是等源码修复再编译吧
> 
> 基础的要自己学，饭来了还要人端不可取......

嗯有道理，我尽力而为哈

Jejz168, 2024-09-14:
大雕已经退回gcc11了😂

zxlhhyccc, 2024-09-14:
> 嗯，大佬，你的办法看上去不错，可我是action编译的，很多Linux指令还不太会，还是等源码修复再编译吧

action加上一行代码即可。。。。
wget -P ./feeds/packages/utils/tini/patches/  https://raw.githubusercontent.com/zxlhhyccc/packages/refs/heads/patch-5/utils/tini/patches/002-Support-POSIX-basename-from-musl-libc.patch

zxlhhyccc, 2024-09-14:
> 大雕已经退回gcc11了

别乱说。。。

Jejz168, 2024-09-14:
> > 大雕已经退回gcc11了
> 
> 
> 
> 别乱说。。。

你没看更新吗？已经撤回了

zxlhhyccc, 2024-09-14:
> 你没看更新吗？已经撤回了

即使撤回了，这个补丁加上去也不影响。。。

love-dy, 2024-09-14:
> > 嗯，大佬，你的办法看上去不错，可我是action编译的，很多Linux指令还不太会，还是等源码修复再编译吧
> 
> Action加上一行代码即可。。。。 `wget -P ./feeds/packages/utils/tini/patches/ https://raw.githubusercontent.com/zxlhhyccc/packages/refs/heads/patch-5/utils/tini/patches/002-Support-POSIX-basename-from-musl-libc.patch`

这个好，我试试

love-dy, 2024-09-14:
[image]
😓又出现了新的错误

zxlhhyccc, 2024-09-15:
> 又出现了新的错误

你编译的是rockchip？

love-dy, 2024-09-15:
> > 又出现了新的错误
> 
> 你编译的是rockchip？看起来是权限问题，但报错不是很清晰，请贴出详细的log。

大佬，我弄到半夜打算放弃，把log也删了，只能重新编译，有log就贴出来

love-dy, 2024-09-15:
> > 又出现了新的错误
> 
> 你编译的是rockchip？看起来是权限问题，但报错不是很清晰，请贴出详细的log。

[image]
[0_config _ core.txt](https://github.com/user-attachments/files/17004647/0_config._.core.txt)

love-dy, 2024-09-15:
刚发现昨天晚上 x86居然编译成功了，看来是r4s里添加了一些不该有的组件

Jejz168, 2024-09-15:
> 刚发现昨天晚上 x86居然编译成功了，看来是r4s里添加了一些不该有的组件

取消kmod-r8125 或者 kmod-r8125-rss 其中的一个
取消kmod-r8126 或者 kmod-r8126-rss 其中的一个
两个冲突，根据你的需求取消一个

love-dy, 2024-09-15:
> > 刚发现昨天晚上 x86居然编译成功了，看来是r4s里添加了一些不该有的组件
> 
> 取消kmod-r8125 或者 kmod-r8125-rss 其中的一个 取消kmod-r8126 或者 kmod-r8126-rss 其中的一个 两个冲突，根据你的需求取消一个

只添加一个 kmod-usb-net-rtl8150  和 kmod-usb-net-rtl8152-vendor，要不我取消一个试试

love-dy, 2024-09-15:
大佬修复源码了，今天能成功编译了

closed 2024-09-15 by binge8