coolsnowwolf/lede #12468: kmod-r8125和kmod-r8125-rss冲突

binge8, 2024-09-15:
### 详细叙述

.config里只有kmod-r8125，没有kmod-r8125-rss啊

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x64

### 详细日志

~~~bash
Configuring uci.
Collected errors:
Configuring grub2.
 * check_data_file_clashes: Package kmod-r8125-rss wants to install file /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/root-x86/lib/modules/6.6.50/r8125.ko
	But that file is already provided by package  * kmod-r8125
 * opkg_install_cmd: Cannot install package kmod-r8125-rss.
 * check_data_file_clashes: Package kmod-r8126-rss wants to install file /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/root-x86/lib/modules/6.6.50/r8126.ko
	But that file is already provided by package  * kmod-r8126
 * opkg_install_cmd: Cannot install package kmod-r8126-rss.
 * resolve_conffiles: Existing conffile /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/root-x86/etc/config/socat is different from the conffile in the new package. The new conffile will be placed at /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/root-x86/etc/config/socat-opkg.
make[2]: *** [package/Makefile:70: package/install] Error 255
~~~

aiamadeus, 2024-09-15:
删掉 .config 重新选。。不要两个都选

binge8, 2024-09-15:
> 删掉 .config 重新选。。不要两个都选

我上面说了啊，.config里只有r8125，没有r8125-rss

closed 2024-09-15 by binge8