coolsnowwolf/lede #12469: 添加luci-app-mosdns编译冲突有哪位大佬知道是什么问题

xueyebai, 2024-09-15:
### 详细叙述

如题，没有添加这个插件的情况下是没问题的！添加了之后和ppp-mod-pppoe冲突，编译不了，新手不会分析问题

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

Configuring ppp-mod-pppoe.
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:111: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

shmilyx, 2024-09-17:
不支持了。

shmilyx, 2024-09-17:
https://github.com/sbwml/luci-app-mosdns/issues/252#issuecomment-2336829767

lilaoye88, 2024-09-22:
不要编译。刷好新固件后手工安装5.3.1

coolsnowwolf, 2026-09-05:
已确认原问题来自额外加入的 `sbwml/luci-app-mosdns` 第三方 feed。该项目维护者在链接的评论中明确说明，当时的新版本不支持 LEDE/Lean；末尾的 `Configuring ppp-mod-pppoe` 只是并行构建输出，不是实际冲突原因，所附日志也遗漏了首个错误。

当前 LEDE 已内置适配本树的 MosDNS，不要再叠加同名第三方 feed。我在当前 x86_64 配置上临时启用并用 `-j30` 实际编译通过，产物为：

- `mosdns_5.3.4-1_x86_64.ipk`（SHA256 `370f7ea7f5bbdcf480b4a13d5f020801c60cf42e9fde369a1c3acdfb35181b6f`）
- `luci-app-mosdns_1.6.6-r1_all.ipk`（SHA256 `adfab201915fcc245890ae3f93b84c481e1e4f6dc614cdcea694473e2aa0c47a`）

因此当前仓库不存在所述 `ppp-mod-pppoe` 编译冲突，无需代码改动，按已解决关闭。

closed 2026-09-05 by coolsnowwolf