coolsnowwolf/lede #12472: 最近编译X86固件发现网络wan和wan6设置界面下面出现DHCP服务器设置项

hztbhtbh, 2024-09-16:
### 详细叙述

最近编译X86固件发现网络wan和wan6设置界面下面出现DHCP服务器设置项，如下图红色框，以前红色框都没有的，这个红色框都不需要设置，有没有办法在编译的时候取消了不编译呢？请各位大佬指点迷津，谢谢 ！
[image]
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86_64软路由

### 详细日志

没有问题日志，请看截图谢谢~

aiamadeus, 2024-09-16:
https://github.com/coolsnowwolf/luci/commit/dd457f9d1ad8ff3cf0a593886e8b66910790ad07 revert 掉

hztbhtbh, 2024-09-16:
> [coolsnowwolf/luci@dd457f9](https://github.com/coolsnowwolf/luci/commit/dd457f9d1ad8ff3cf0a593886e8b66910790ad07) revert 掉

感谢大佬的回复，已经编译去掉了，谢谢~

Kisoul, 2024-09-17:
> > [coolsnowwolf/luci@dd457f9](https://github.com/coolsnowwolf/luci/commit/dd457f9d1ad8ff3cf0a593886e8b66910790ad07) revert 掉
> 
> 感谢大佬的回复，已经编译去掉了，谢谢~

具体怎么办？麻烦解答一下

closed 2024-11-20 by aiamadeus