coolsnowwolf/lede #12475: 编译问题

gssiwmd, 2024-09-16:
### 详细叙述

macos安装虚拟机ubuntu20.04 编译错误

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

MacBook air 15 M3芯片

### 详细日志

install -d -m0755 /home/gswangmd/lede/bin/packages/x86_64/gssiwmdpackages
/home/gswangmd/lede/staging_dir/host/bin/fakeroot /home/gswangmd/lede/staging_dir/host/bin/bash /home/gswangmd/lede/scripts/ipkg-build -m "" /home/gswangmd/lede/build_dir/target-x86_64_musl/dns2tcp-1.1.2/ipkg-x86_64/dns2tcp /home/gswangmd/lede/bin/packages/x86_64/gssiwmdpackages
Packaged contents of /home/gswangmd/lede/build_dir/target-x86_64_musl/dns2tcp-1.1.2/ipkg-x86_64/dns2tcp into /home/gswangmd/lede/bin/packages/x86_64/gssiwmdpackages/dns2tcp_1.1.2-2_x86_64.ipk
echo "dns2tcp" >> /home/gswangmd/lede/staging_dir/target-x86_64_musl/pkginfo/dns2tcp.default.install
make[3]: Leaving directory '/home/gswangmd/lede/feeds/gssiwmdpackages/dns2tcp'
time: package/feeds/gssiwmdpackages/dns2tcp/compile#0.24#0.01#0.26
make[3]: Entering directory '/home/gswangmd/lede/feeds/packages/lang/golang/golang'
Makefile:431: *** go-bootstrap cannot be installed on linux/arm64.  Stop.
make[3]: Leaving directory '/home/gswangmd/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#0.03#0.00#0.04
    ERROR: package/feeds/packages/golang [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/golang/host/compile] Error 1
make[2]: Leaving directory '/home/gswangmd/lede'
make[1]: *** [package/Makefile:110: /home/gswangmd/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gswangmd/lede'
make: *** [/home/gswangmd/lede/include/toplevel.mk:231: world] Error 2
gswangmd@wmdmacubuntuserver:~/lede$

aiamadeus, 2024-09-17:
go 工具链不支持 arm64 上编译

yanxin152133, 2024-11-18:
Makefile:431: *** go-bootstrap cannot be installed on linux/arm64. Stop.
https://blog.hentioe.dev/posts/arm64-build-openwrt.html
[image]

coolsnowwolf, 2026-09-05:
已按当前代码验证。这个错误来自 Go 1.4 bootstrap 本身不支持 `linux/arm64`，不是目标平台选 x86 导致的。当前 golang 包已经提供外部 bootstrap 配置，不应把 `linux/arm64` 直接加入旧 bootstrap 的支持列表。

在 ARM64 Ubuntu 中先安装一个可运行的宿主机 Go，然后执行：

```sh
go env GOROOT
./scripts/config --set-str GOLANG_EXTERNAL_BOOTSTRAP_ROOT "$(go env GOROOT)"
make defconfig
```

也可在 `make menuconfig` 的 Languages → Go → Configuration 中设置 `External bootstrap Go root directory`。

我用当前 Makefile 模拟 `linux/arm64` 做了解析验证：未设置时可复现 `go-bootstrap cannot be installed on linux/arm64`；设置外部 GOROOT 后解析成功，并正确生成后续各级 Go bootstrap 构建链。由于当前树已有正规解决入口，无需修改软件包，按已解决关闭。

closed 2026-09-05 by coolsnowwolf