coolsnowwolf/lede #12477: R5S设备固件其中一LAN口无法识别

lysgwl, 2024-09-17:
### 详细叙述

lede版本编译的r5s设备固件，两个LAN口其中一个LAN口无法识别，中间的2.5GLAN口无法识别，另外一个可以使用，查询使用的驱动是R8125驱动，什么原因呢？

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

friendlywrt r5s

### 详细日志

以下时内核日志：
[ 0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[ 0.000000] Linux version 5.15.162 (root@ubuntu22) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r27917-81a1f98d5b) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP PREEMPT Thu Jul 18 06:35:54 2024
[ 0.000000] Machine model: FriendlyElec NanoPi R5S
[ 0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
[ 0.000000] printk: bootconsole [uart8250] enabled
[ 0.000000] Zone ranges:
[ 0.000000] DMA [mem 0x0000000000200000-0x00000000efffffff]
[ 0.000000] DMA32 empty
[ 0.000000] Normal empty
[ 0.000000] Movable zone start for each node
[ 0.000000] Early memory node ranges
[ 0.000000] node 0: [mem 0x0000000000200000-0x00000000efffffff]
[ 0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000000efffffff]
[ 0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[ 0.000000] cma: Reserved 16 MiB at 0x00000000ee800000
[ 0.000000] psci: probing for conduit method from DT.
[ 0.000000] psci: PSCIv1.1 detected in firmware.
[ 0.000000] psci: Using standard PSCI v0.2 function IDs
[ 0.000000] psci: MIGRATE_INFO_TYPE not supported.
[ 0.000000] psci: SMC Calling Convention v1.2
[ 0.000000] percpu: Embedded 18 pages/cpu s36504 r8192 d29032 u73728
[ 0.000000] pcpu-alloc: s36504 r8192 d29032 u73728 alloc=184096
[ 0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[ 0.000000] Detected VIPT I-cache on CPU0
[ 0.000000] CPU features: detected: GIC system register CPU interface
[ 0.000000] CPU features: detected: Virtualization Host Extensions
[ 0.000000] CPU features: detected: Hardware dirty bit management
[ 0.000000] alternatives: patching kernel code
[ 0.000000] Built 1 zonelists, mobility grouping on. Total pages: 967176
[ 0.000000] Kernel command line: coherent_pool=2M console=ttyS2,1500000 console=tty1 earlycon=uart8250,mmio32,0xfe660000 root=PARTUUID=5452574f-02 rw rootwait
[ 0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[ 0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[ 0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[ 0.000000] Memory: 3821484K/3930112K available (8768K kernel code, 2380K rwdata, 2568K rodata, 1664K init, 734K bss, 92244K reserved, 16384K cma-reserved)
[ 0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[ 0.000000] rcu: Preemptible hierarchical RCU implementation.
[ 0.000000] rcu: RCU event tracing is enabled.
[ 0.000000] rcu: RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[ 0.000000] Trampoline variant of Tasks RCU enabled.
[ 0.000000] Tracing variant of Tasks RCU enabled.
[ 0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[ 0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[ 0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[ 0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[ 0.000000] GICv3: 320 SPIs implemented
[ 0.000000] GICv3: 0 Extended SPIs implemented
[ 0.000000] GICv3: Distributor has no Range Selector support
[ 0.000000] Root IRQ handler: gic_handle_irq
[ 0.000000] GICv3: 16 PPIs implemented
[ 0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fd460000
[ 0.000000] ITS [mem 0xfd440000-0xfd45ffff]
[ 0.000000] ITS@0x00000000fd440000: allocated 8192 Devices @440000 (indirect, esz 8, psz 64K, shr 0)
[ 0.000000] ITS@0x00000000fd440000: allocated 32768 Interrupt Collections @450000 (flat, esz 2, psz 64K, shr 0)
[ 0.000000] ITS: using cache flushing for cmd queue
[ 0.000000] GICv3: using LPI property table @0x0000000000460000
[... 424 lines cut ...]
[ 16.171644] xt_FULLCONENAT: Copyright (C) 2018 Chion Tang [tech@chionlab.moe](mailto:tech@chionlab.moe)
[ 16.174143] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[ 16.721851] eth1: 0xffff800009fd0000, aa:ae:b5:02:67:82, IRQ 76
[ 16.769755] br-lan: port 1(eth1) entered blocking state
[ 16.770255] br-lan: port 1(eth1) entered disabled state
[ 16.771093] device eth1 entered promiscuous mode
[ 16.778417] rk_gmac-dwmac fe2a0000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 16.868999] rk_gmac-dwmac fe2a0000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211F Gigabit Ethernet] (irq=POLL)
[ 16.876944] dwmac4: Master AXI performs any burst length
[ 16.877465] rk_gmac-dwmac fe2a0000.ethernet eth0: No Safety Features support found
[ 16.878164] rk_gmac-dwmac fe2a0000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[ 16.879059] rk_gmac-dwmac fe2a0000.ethernet eth0: registered PTP clock
[ 16.880043] rk_gmac-dwmac fe2a0000.ethernet eth0: configuring for phy/rgmii link mode
[ 16.947012] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[ 18.089631] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[ 40.490818] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[ 81.928990] r8125: eth1: link up
[ 81.929371] br-lan: port 1(eth1) entered blocking state
[ 81.929851] br-lan: port 1(eth1) entered forwarding state
[ 81.930705] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[ 83.472956] r8125: eth1: link down
[ 83.490155] br-lan: port 1(eth1) entered disabled state
[ 85.581005] r8125: eth1: link up
[ 85.581380] br-lan: port 1(eth1) entered blocking state
[ 85.581860] br-lan: port 1(eth1) entered forwarding state

aiamadeus, 2024-09-18:
friendlywrt 都直接打过来了？

lysgwl, 2024-09-18:
> friendlywrt 都直接打过来了？

额额。。。知错了，拷贝赋值的。这个不是重点。

aiamadeus, 2024-09-18:
你这内核都不是 lede

lysgwl, 2024-09-18:
> 你这内核都不是 lede

已经更正了，可能是传错了。现在问题依然存在，lan的eth2还是无法被识别，还有lan的指示灯状态显示也是错误的。

closed 2024-11-16 by lysgwl