coolsnowwolf/lede #12478: 系统启动后docker报错

love-dy, 2024-09-17:
### 详细叙述

[    9.004365] br-lan: port 1(eth0) entered blocking state
[    9.004591] br-lan: port 1(eth0) entered forwarding state
[   16.387873] overlay: filesystem on /opt/docker/check-overlayfs-support1396556380/upper not supported as upperdir
[   16.624817] traps: check[5381] general protection fault ip:8049007 sp:ffc66ef0 error:402 in check[8049000+1000]
[  351.845628] overlay: filesystem on /opt/docker/check-overlayfs-support31793842/upper not supported as upperdir
[  352.068112] traps: check[9214] general protection fault ip:8049007 sp:ffd7a7d0 error:402 in check[8049000+1000]

大佬求解，每次docker启动都会报这两个错误
https://github.com/coolsnowwolf/lede/issues/8698
用的esxi 

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

none

love-dy, 2024-09-18:
刚试了r4s, 系统启动后内核日志里也有这个错误
overlay: filesystem on /opt/docker/check-overlayfs-support31793842/upper not supported as upperdir

不知道是不是问题，据我分析，每次系统或docker重启的时候会报错，运行过程中好像没有报错提示，这应该是docker初始化阶段系统没有识别，启动完成后就正常了，不知道分析的对不对，大佬可以分析分析吗

coolsnowwolf, 2024-09-19:
这个不是报错，是告诉你，你的 /opt 目录不能是根目录 overlay fs

love-dy, 2024-09-19:
> 这个不是报错，是告诉你，你的 /opt 目录不能是根目录 overlay fs

不是错误就好，那我就放心了

coolsnowwolf, 2026-09-05:
已确认这是 dockerd 启动时探测 overlayfs 存储驱动能力产生的日志：当 `/opt/docker` 所在文件系统本身是根 overlayfs 时，不能再直接作为另一个 overlayfs 的 upperdir。Docker 会据此选择可用的存储方式；它不表示内核启动失败。维护者已经在上方说明，报告者也确认无需继续处理，因此按已解决关闭。若实际容器无法启动，请另附 `docker info` 中的 `Storage Driver`、`mount` 和完整 dockerd 日志重新报告。

closed 2026-09-05 by coolsnowwolf