coolsnowwolf/lede #12479: pgyvpn编译报错

1105384898, 2024-09-18:
### 详细叙述

Package pgyvpn is missing dependencies for the following libraries:
libcrypto.so.1.1
libssl.so.1.1
make[3]: *** [Makefile:133: /home/yangyang/lede/bin/targets/x86/64/packages/pgyvpn_3.1.0-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/yangyang/lede/feeds/packages/net/pgyvpn'
time: package/feeds/packages/pgyvpn/compile#0.10#0.22#0.55
    ERROR: package/feeds/packages/pgyvpn failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/pgyvpn/compile] Error 1
make[2]: Leaving directory '/home/yangyang/lede'
make[1]: *** [package/Makefile:110: /home/yangyang/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yangyang/lede'
make: *** [/home/yangyang/lede/include/toplevel.mk:231：world] 错误 2

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

Package pgyvpn is missing dependencies for the following libraries:
libcrypto.so.1.1
libssl.so.1.1
make[3]: *** [Makefile:133: /home/yangyang/lede/bin/targets/x86/64/packages/pgyvpn_3.1.0-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/yangyang/lede/feeds/packages/net/pgyvpn'
time: package/feeds/packages/pgyvpn/compile#0.10#0.22#0.55
    ERROR: package/feeds/packages/pgyvpn failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/pgyvpn/compile] Error 1
make[2]: Leaving directory '/home/yangyang/lede'
make[1]: *** [package/Makefile:110: /home/yangyang/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yangyang/lede'
make: *** [/home/yangyang/lede/include/toplevel.mk:231：world] 错误 2

mygithub1981, 2024-10-08:
我也有同样问题，楼主怎么解决的？

sKyissKy, 2024-10-28:
问题又复现了，这个怎么搞，有大佬能在issue留个手工解决办法？